robotology/icub-models #1: iCubNancy01 urdf

InriaBrice, 2017-06-30:
Hi !

We need some features about iCubNancy01 urdf to convert urdf to sdf file with gz sdf -p command : 

Links can't have masse = 0,
Links can't have inertia matrix diagonal null -> ixx, iyy, izz != 0,
It happens for link like wrist or hip_yaw, so it's not an end effector (we can't just remove this link otherwise hand can't be here ),

We need a version without dh frame because we just don't use them,

About Center of masses of links, soemtimes it seems like they are not at the right position (especially in the arms they are a bit forward ).

[image]

Also if it is possible to have all link frame with X axis forward and Z axis upward.

Also i was asking myself, why meshes are not Centered at origin in .dae files ? ( Also it's possible to set frame meshes to get X forward and Z upward ) .

I think link wrist could be more in the middle of the forearm, i mean at the middle of end of the forearm, they are actually ahead and in the forearm.

Also it would be nice if iCub was turned forward ( looking at +X axis ), without turn robot in gazebo or whatever (just with urdf spawn )

For now i customized all to get something like expected, but some inertia matrix are making robot a bit instable.

traversaro, 2017-07-04:
Hi @InriaBrice , moving the issue over to `icub-model-generator` where we already have all the other discussions on the generated models.

traversaro, 2017-07-04:
Issue moved to [robotology-playground/icub-model-generator #50](https://github.com/robotology-playground/icub-model-generator/issues/50) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-07-04 by traversaro