robotology/icub-models #5: Use the models in Gazebo

S-Dafarra, 2017-10-23:
Can you provide some instructions on how to use these models in Gazebo?
For example, I added the ``/icub-models-install/share/iCub/robots`` folder to the ``GAZEBO_MODEL_PATH`` variable, but Gazebo is not able to find the meshes. What is the best way to load the meshes without copying their folder into ``/icub-models-install/share/iCub/robots``?
Can I run all the controllers as with [``icub-gazebo``](https://github.com/robotology/icub-gazebo) with no problems?

@fiorisi @traversaro @nunoguedelha @gabrielenava

traversaro, 2017-10-23:
The relevant issues are: 
 * https://github.com/robotology-playground/icub-models/pull/3 
 * https://github.com/robotology-playground/icub-model-generator/issues/52 
I don't know what is currently working or not, probably @diegoferigo can comment on this.

S-Dafarra, 2017-10-23:
Also I am a bit confused, what's the relation between this repo and [``icub-model-generator``](https://github.com/robotology-playground/icub-model-generator)? 
Because I am noticing the same issue about the meshes that should have been solved with [this PR](https://github.com/robotology-playground/icub-model-generator/pull/49). 
Should I wait the checklist of https://github.com/robotology-playground/icub-model-generator/issues/52 to be completed before having a running a simulation?

cc. @DanielePucci

traversaro, 2017-10-23:
> Also I am a bit confused, what's the relation between this repo and icub-model-generator?

The content of this repo are automatically generated from the software at icub-model-generator using a Travis job. Please see https://github.com/robotology-playground/icub-model-generator/blob/master/.travis.yml#L76 for the logic.

S-Dafarra, 2017-10-24:
Thanks! Do you have any idea about how to make Gazebo finding the meshes? Because I tried to add the ``/icub-models-install/share/iCub`` folder to the ``GAZEBO_MODEL_PATH`` variable but it is not working.

traversaro, 2017-10-24:
Relevant issue: https://github.com/robotology-playground/icub-model-generator/issues/25#issuecomment-310343556 .

diegoferigo, 2017-10-24:
For future reference I recommended f2f to try adding the folders listed [in this file](https://github.com/diegoferigo/dockerfiles/blob/4a65fb79275e45f20cecfa939f32dced7abe2fd3/Development/Dockerfile#L279). The README should be updated with Gazebo instructions.

S-Dafarra, 2017-10-24:
Thanks! Indeed the folder to be added in ``GAZEBO_MODEL_PATH`` was ``icub-models-install/share``.

nunoguedelha, 2017-10-24:
Regarding issue https://github.com/robotology-playground/icub-model-generator/issues/52, I've added some information https://github.com/robotology-playground/icub-model-generator/issues/52#issuecomment-338957495 for helping @fiorisi solving the last point in the checklist (small inertia on Gazebo).

S-Dafarra, 2017-10-24:
I modified the installed URDF by hand as @nunoguedelha suggested, but Gazebo **crashes**. From a rapid debug with @francesco-romano it seems that these lines are causing the crash:
```
<plugin name="iCub_yarp_gazebo_plugin_ACCpart" filename="libgazebo_yarp_distributedinertials.so">
  <yarpConfigurationFile>model://iCub/conf/MTB/gazebo_icub_left_leg_mtb.ini</yarpConfigurationFile>
</plugin>
<plugin name="iCub_yarp_gazebo_plugin_ACCpart" filename="libgazebo_yarp_distributedinertials.so">
  <yarpConfigurationFile>model://iCub/conf/MTB/gazebo_icub_right_leg_mtb.ini</yarpConfigurationFile>
</plugin>
<plugin name="iCub_yarp_gazebo_plugin_ACCpart" filename="libgazebo_yarp_distributedinertials.so">
  <yarpConfigurationFile>model://iCub/conf/MTB/gazebo_icub_left_arm_mtb.ini</yarpConfigurationFile>
</plugin>
<plugin name="iCub_yarp_gazebo_plugin_ACCpart" filename="libgazebo_yarp_distributedinertials.so">
  <yarpConfigurationFile>model://iCub/conf/MTB/gazebo_icub_right_arm_mtb.ini</yarpConfigurationFile>
</plugin>
<plugin name="iCub_yarp_gazebo_plugin_ACCpart" filename="libgazebo_yarp_distributedinertials.so">
  <yarpConfigurationFile>model://iCub/conf/MTB/gazebo_icub_torso_mtb.ini</yarpConfigurationFile>
</plugin>
```
Maybe I need to use some particular branches?

In addition it seems that the left arm has some problems
[image]

nunoguedelha, 2017-10-24:
@S-Dafarra these plugins have been released in `gazebo-yarp-plugins` but I don't remember if it was in `devel` or `master` branch. Let me check.
Could you check if you have the inertialMTB boards plugins? look for the string **"11b1"** for instance. We can keep on by chat.

S-Dafarra, 2017-10-24:
I am on the ``devel`` branch of ``gazebo-yarp-plugins``.

S-Dafarra, 2017-10-25:
Closing this issue since all the above problems have been collected in https://github.com/robotology-playground/icub-models/issues/7.

closed 2017-10-25 by S-Dafarra