robotology/icub-models #6: Update README to use models with Gazebo

S-Dafarra, 2017-10-24:
Following https://github.com/robotology-playground/icub-models/issues/5, I think we could add few lines to the [README](https://github.com/robotology-playground/icub-models/blob/master/README.md).

- Line 3, 
>Repository containing models [automatically](https://github.com/robotology-playground/icub-models/issues/5) generated by https://github.com/robotology-playground/icub-model-generator .

- Somewhere at the end:
> ## Gazebo
> In order to use these models in Gazebo, set up the simulation environment following the instructions provided in the [icub-gazebo](https://github.com/robotology/icub-gazebo) repository, and add the following line to your ``.bashrc``:
>``
>export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/your_installation_folder/share/iCub/robots:/your_installation_folder/share
>``   

@diegoferigo feel free to edit it as you prefer :blush:

traversaro, 2017-10-24:
👍 , feel free to open a PR.

diegoferigo, 2017-10-25:
@S-Dafarra I edited the README accordingly, can you please check all is working with the given instructions?

I have some doubts on the very last entry of `GAZEBO_MODEL_PATH`, I'm not sure it is required.

commit 9a7bd8ef (2017-10-25): Update README.md

Fixes #6
commit 9219c812 (2017-10-25): Update README.md

Fixes #6
closed 2017-10-25 by diegoferigo