robotology/icub-models #7: Problems when using these models in Gazebo

S-Dafarra, 2017-10-25:
While using the ``icub-models`` in Gazebo I encountered the following problems:
- [ ] The inertias of some links are too small and need to be modified by hand as suggested in [this comment](https://github.com/robotology-playground/icub-model-generator/issues/52#issuecomment-338957495).
- [ ] When the robot is inserted in Gazebo, it is "submerged" in the ground and needs to be moved by hand pausing Gazebo beforehand.
- [ ] Once I press play on Gazebo with the robot correctly placed, Gazebo crashes. See [this comment](https://github.com/robotology-playground/icub-models/issues/5#issuecomment-339034743)
- [ ] Both the wrist yaw joints go in hardware fault. Apparently the position control is unstable and it brings the joints toward the limits. I tried to reduce the gains, but it did not work. 

Cc.
@nunoguedelha @traversaro @diegoferigo @francesco-romano @fiorisi @DanielePucci

traversaro, 2017-10-25:
* [ ] The meshes in the foot should be substituted with a simpler primitive shapes (box or four spheres)

traversaro, 2017-10-25:
> Both the wrist yaw joints go in hardware fault. Apparently the position control is unstable and it brings the joints toward the limits. I tried to reduce the gains, but it did not work. 

With @nunoguedelha we looked into computing gazebo gains nominally equivalent to the real robot gains. I think we should set the gain on the Gazebo models that are approximately equivalent to the real robot one.

S-Dafarra, 2017-10-25:
@traversaro I tried to reduce them a lot. The only difference was that the joints did not go into HF, but they reached the same limit position.

traversaro, 2017-10-25:
> @traversaro I tried to reduce them a lot. The only difference was that the joints did not go into HF, but they reached the same limit position.

Mhh. This does not seem just an gains issue.

diegoferigo, 2017-10-25:
> When the robot is inserted in Gazebo, it is "submerged" in the ground and needs to be moved by hand pausing Gazebo beforehand

I thought that `icub-models` should contain only urdfs / meshes, and then [`icub-gazebo-wholebody`](https://github.com/robotology-playground/icub-gazebo-wholebody) is required for the configuration and placement of these models.

traversaro, 2017-10-25:
> I thought that icub-models should contain only urdfs / meshes, and then icub-gazebo-wholebody is required for the configuration and placement of these models.

In theory icub-gazebo-wholebody was meant to be just a repository containing DIC/LOC2 specific stuff, while icub-gazebo and icub-models should be usable indipendently (as usual, all this decision can be changed). 
The offset of the robot was classically introduced in the converted SDF model by specifying explicitly the pose of the model (https://github.com/robotology/icub-gazebo/blob/master/icub/icub.sdf#L4). In some other projects distributing Gazebo models in URDF form, the strategy that I have used is to keep two Gazebo models: one without any YARP plugin, and a composite one that includes the raw one, adds the YARP plugins and the necessary offset. Unfortunately I don't have any example of this in public repositories.

traversaro, 2017-10-25:
>  Once I press play on Gazebo with the robot correctly placed, Gazebo crashes. See this comment
This seems to be related to https://github.com/robotology-playground/icub-model-generator/issues/54 .

traversaro, 2017-10-25:
> The inertias of some links are too small and need to be modified by hand as suggested in this comment.
Relative issue: https://github.com/robotology/icub-gazebo/issues/22 .

traversaro, 2017-10-25:
I would also add to the problems that we should install configuration files only for `iCubGazeboV2_5`, as all other models are expected not to work.

traversaro, 2017-10-25:
For some of this issue, opening a separate thread could improve the readability of the issue.

S-Dafarra, 2017-10-25:
Maybe this issue should become an epic

nunoguedelha, 2017-11-01:
@S-Dafarra @traversaro I've edited the first two comments of the issue for checking the solved problems. Remaining problems are also tracked in the issue that was opened in July and was tracking the work to be done for using the generated models on Gazebo (turned now into an Epic): [icub-model-generator/#52](https://github.com/robotology-playground/icub-model-generator/issues/52).

S-Dafarra, 2017-11-02:
Thanks @nunoguedelha!! Maybe we can close this issue and point to https://github.com/robotology-playground/icub-model-generator/issues/52 in the [README](https://github.com/robotology-playground/icub-models/blob/master/README.md) of this repo to avoid duplicates. 

Can I also ask you some more details about
>not a real issue, was due to wrong environment variable setting (GAZEBO_MODEL_PATH)

Which folder should be added? Was it a problem in the placement of the configuration files?

nunoguedelha, 2017-11-02:
Edited the original comment of the issue regarding the third listed problem.

nunoguedelha, 2018-01-15:
The new models are now supported on Gazebo and have been tested. Any remaining improvements will continue to be tracked in https://github.com/robotology-playground/icub-model-generator/issues/52.
Closing this issue now.

closed 2018-01-15 by nunoguedelha