robotology/icub-models #13: iCubGazeboV2_5 unstable if it is controlled using velocity mode

GiulioRomualdi, 2018-05-07:
When I try to switch from `position` control to `velocity` control iCub jumps and the switched joint goes in `hardware fault`.
Here a simple example.

[image]

I have tried to tune the PID controller of the l_shoulder_pitch` joint:

| Gain | Old   |  New |
|:----:|-------|:----:|
|  Kp  | 8.726 | 0.85 |
|  Kd  | 0.035 |   0  |
|  Ki  | 0.002 |  0.15 |

This is the result 

[image]

traversaro, 2018-05-07:
Great @GiulioRomualdi . You can open a PR against the configuration file that are stored in icub-model-generator, see : https://github.com/robotology-playground/icub-model-generator/tree/master/simmechanics/data/icub2_5/conf .

fiorisi, 2018-05-08:
With @gabrielenava we noticed that in the new `gif` the arm of the robot is slowly moving down. this is probably due to the fact that the Kp gain is low. Did you try to reduce the time step of the Gazebo simulation with the [`Old`](https://github.com/robotology-playground/icub-models/issues/13#issue-320776453) gains?

GiulioRomualdi, 2018-05-08:
@fiorisi I have just tried. The results is the same of before: the switched joint goes in hardware fault.

traversaro, 2020-11-12:
With PR https://github.com/robotology/icub-models-generator/pull/138, we switched the velocity control mode to actually send a velocity to a low-level integrator, whose output is then send to the usual position control. For this reason, I think that this issue is now solved, and it can be closed. Feel free to reopen if a similar error is still present.

closed 2020-11-12 by traversaro