robotology/icub-models #20: Document frames and link frames situation

traversaro, 2019-02-08:
*  Document that the location of the link frames is not consistent between the models, and that for simmechanics models only the link frames listed in https://github.com/robotology/icub-model-generator/blob/52ee80fdbfc588e09010dd418fd5904c6ac7383c/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in#L552 have a strictly defined location 
* Move documentation from http://wiki.icub.org/wiki/ICub_Model_naming_conventions to this repo

traversaro, 2019-02-08:
cc @GiulioRomualdi @S-Dafarra

Yeshasvitvs, 2019-02-19:
@kouroshD

Yeshasvitvs, 2019-02-28:
During the retargeting work, @kouroshD identified the link frames of the robot. The following figure shows the link frames of the icub frames but for our retargeting work/paper they are represented as the frames attached at the center of each of the links.

[image]

traversaro, 2019-02-28:
Note that, as mentioned in https://github.com/robotology/icub-models/issues/20#issue-408154313, the frames of all the links that are not part of the legs can change at arbitrary times any time that the CAD model is generated again from Creo.