robotology/icub-models #21: ICub default posture

cmastalli, 2019-04-17:
Hi guys,

I was looking for the joint values for a default posture. I meant the posture that the robots has in when it's standing statically.

Grazie mille,
Carlos

traversaro, 2019-04-18:
Hi @cmastalli ! 
It depends on the specific model of iCub, but for the `iCubGazeboV2_5` stored in this repo the position that we typically use for balancing experiments `TwoFeetStanding` position stored in https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/app/robots/iCubGazeboV2_5/homePoseBalancing.ini#L20 or the `YogaPP` position stored in the same file: https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/app/robots/iCubGazeboV2_5/homePoseBalancing.ini#L6 .

@gabrielenava can add more information if necessary, and feel free to ask more clarifications.

cmastalli, 2019-04-18:
@traversaro grazie mille!

closed 2019-04-18 by cmastalli