robotology/icub-models #22: Clarification on the FT frames in the FT CAD model

gabrielenava, 2019-05-21:
Looking at the CAD model of the FT sensors, more specifically, `ic_007_a_001.asm`, for the FT sensor with M3 skrews, I noted that the `SCSYS_FT45` frame, which should be the **sensor frame** of the FT sensor, is reversed w.r.t. the `sketch` that represents the positive **x** and **y** axis of the sensor's frame. See the pictures below:

[image]
[image]

@fiorisi am I missing something here? Which frame should I follow when mounting the FT?

traversaro, 2019-05-21:
For reference, the FT sensor frame is documented in http://wiki.icub.org/wiki/FT_sensor .

gabrielenava, 2019-05-21:
Mmh, comparing the wiki and the CAD it seems that the `sketch` is  not the sensor frame, as in the wiki the **x** axis is pointing in the direction opposite to the one where there is the **hole for the cable**, but this is not the case in the CAD model:

[image]
[image]

But the models are not the same, as the skrews pattern placement is different. I'll follow the wiki convention for now.

traversaro, 2019-05-21:
As the `SCSYS_FT45` and the wiki definition are consistent, I would say that those sketch are misleading, but @fiorisi can probably provide more insight. 

A possible explanation of those strange directions in the sketch is that a tricky part of defining FT sensors measures is that defining the **coordinate frame** in which the force and torque are expressed is not sufficient, you also need to specify if the measured force is the one that the "top" of the sensor is exerting on the "bottom", or the one that the "bottom" is exerting on the "top". In SDF, this characteristic is called `measure_direction`. See https://bitbucket.org/osrf/gazebo/issues/940/force-torque-sensor-forcetorquesensorcc, https://github.com/robotology/gazebo-yarp-plugins/issues/73 and http://gazebosim.org/tutorials?tut=force_torque_sensor&cat=sensors for more info on this. 

An important thing to note is that as long as those two forces are expressed in the same frame, the only difference is in their sign by the Newton's third law ( https://en.wikipedia.org/wiki/Newton%27s_laws_of_motion#Newton%27s_third_law ). However it is extremely common, especially is someone is considering  only two components of the measure, to mistake a different `measure_direction` of the FT sensor for the fact that the measure is expressed in a different reference frame, that is actually only possible if you consider left handed reference frames.

fiorisi, 2019-05-22:
Thanks @gabrielenava for checking the frame in this model.
Please note that anyhow the assembly `ic_007_a_001.asm` refers to the FT model `Mk3`. 
In your design please use the `Mk4` which corresponds to the assembly `ic_008_a_001.asm`.
In this assembly the frame was corrected when we used the sensor in the [AnDy sandals CAD model](https://github.com/dic-iit/element_ftSk_Shoes/issues/71).

[image]

cc @miggia

gabrielenava, 2019-05-25:
Thanks @fiorisi, the `ic_008_a_001.asm` is coherent with the wiki and sketch and sensor frame have the same orientation. I changed the FT sensor I amusing in the CAD. Closing the issue.

closed 2019-05-25 by gabrielenava