robotology/icub-models #28: Add the possibility to use multipleanalogsensorsclient and multipleanalogsensorsremapper also in Gazebo

S-Dafarra, 2020-03-12:
Recently, on the robot, we started using the ``multipleanalogsensorremapper``: https://github.com/robotology/robots-configuration/tree/master/iCubGenova04/wrappers/inertials

The corresponding plugin seems also available in ``GazeboYarpPlugins``: https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/imu/include/gazebo/IMU.hh

How should I modify the ``urdf`` and the configuration files in order to have a parallel setup also in Gazebo?

@prashanthr05 @nunoguedelha @traversaro

traversaro, 2020-03-13:
Hi @S-Dafarra, as far as I remember (unfortunately) the plugins in gazebo-yarp-plugins are not ready to simulate more then one IMU exposed in as a single multipleanalogserver device, as it happens on the robot. Recently @Nicogene worked in adding support to the gazebo_yarp_imu device https://github.com/robotology/gazebo-yarp-plugins/pull/443 .

To actually support this case there are different possible routes, the fastest at the moment to me seems to create (or modify starting from either the `gazebo_yarp_inertialMtb` or `gazebo_yarp_inertialMtbPart` plugins) a model plugin called `gazebo_yarp_multipleimu` `gazebo_yarp_distributedinertial` that retrieves the measure of multiple IMU Gazebo sensors, and expose it over the network via `multipleanalogsensorsserver` .  Note that in this case you would not use directly a `multipleanalogsensorsremapper` device, but you would basically emulate it via the configuration file of this `gazebo_yarp_distributedinertial` device, in which you would pass the IMUs to use. 

Other most advanced strategies (but that may be trickier to implement) would create a plugin to reuse the existing gazebo_yarp_imu implementation, and then access the pointer of the existing YARP devices to combine them in a `multipleanalogsensorsremapper` that then publishes its measures via a `multipleanalogsensorsserver`. The most radical example of this strategy is https://github.com/robotology/gazebo-yarp-plugins/issues/391 .

@S-Dafarra I am availabe to discuss about this directly if you think someone can commit work on it.

traversaro, 2021-02-11:
> Other most advanced strategies (but that may be trickier to implement) would create a plugin to reuse the existing gazebo_yarp_imu implementation, and then access the pointer of the existing YARP devices to combine them in a `multipleanalogsensorsremapper` that then publishes its measures via a `multipleanalogsensorsserver`. The most radical example of this strategy is [robotology/gazebo-yarp-plugins#391](https://github.com/robotology/gazebo-yarp-plugins/issues/391) .

https://github.com/robotology/gazebo-yarp-plugins/issues/391 has been solved by https://github.com/robotology/gazebo-yarp-plugins/pull/532 , so now it should be possible to relatively easily reuse the same configuration files used in the real robots, at least for the wrappers.