robotology/icub-models #30: Give official names to the iCub hand joints

kouroshD, 2020-03-13:
Currently, the iCub hand joints does not have common (official) names in order to be addressed, similar to what we have for the actuated axis names in http://wiki.icub.org/wiki/ICub_joints . 
This naming possibly can be common for both the current version of the iCub hands and the new version.

CC @traversaro @fiorisi  @pattacini

traversaro, 2020-07-17:
Another issue in which having proper names for the finger joints is https://github.com/robotology/gazebo-yarp-plugins/issues/476#issuecomment-659930517 .

fiorisi, 2020-07-17:
cc @marcellosavoldi @ale-git

traversaro, 2020-07-17:
To clarify, controlled/user exposed DOFs/axis are given proper names in http://wiki.icub.org/wiki/ICub_joints . What we are missing are: 
* Names for the links that composed the hands 
* Names for physical joints that interconnect the links in the hands

marcellosavoldi, 2020-07-23:
If I understand well ( I'm not sure, sorry)  , we mechanical guys need to rename every time the parts of the original model from CREO and to do this we need to create a new assembly (disconnected from the original) and modify this last.
About the coordinate system is not a problem add in the original CREO file it with the right name and position, but to have a model with the correct name we need to duplicate the original CAD part and renamed all parts ( not easy and not quick)
Do you think will be possible find a solution to keep the original name ( only the name ,not the coordinate system) we use? In this case we can save time and be more quickly to create the model.

... a medal to everyone who understands my english...

ale-git, 2020-07-23:
The problem is the new and old hands are different, and they don't have the same joints.

traversaro, 2020-07-23:
Hi @ale-git and @marcellosavoldi ! 

Just to clarify, this issue is about the original hand of iCub (the one that is still the iCub hand in the latest iCub 2.7 released) and it is just about agreeing on some names for the physical joints. I am not sure I understood @marcellosavoldi point (@fiorisi probably can help on this) but regarding @ale-git comment, I think it is perfectly fine for the iCub Mk3 hand to have different names for the joints and links if they are different, even if it would still be nice to call with the same name links that have a direct analogy between the two links (i.e. it would be great to continue to refer to the palm link as `l_hand` and `r_hand` as done until now).

ale-git, 2020-07-23:
@traversaro ooops... 😊