robotology/icub-models #34: Weight and drawing of hand

KhomutovEvgeniy, 2020-04-01:
What is the weight of hand iCub?

Could you send a link for drawing of hand with sizes?

traversaro, 2020-10-21:
Hi @KhomutovEvgeniy, sorry for the delayed response, I did not noticed this issue back in time. You were able to satisfy your question by other means?

KhomutovEvgeniy, 2020-10-21:
Hi @traversaro, I solved the issue with the drawing of hand, but I could'n find information about weight of forearm and hand, I found information only about mass of the whole arm.

traversaro, 2020-10-22:
The nominal weight of the hand and the forrearm for the iCub 2.5 model is available in the URDF:
* `l_hand` link (from https://github.com/robotology/icub-models/blob/v1.17.0/iCub/robots/iCubGazeboV2_5/model.urdf#L2050):  0.247806 Kg 
* `l_forearm` link (from https://github.com/robotology/icub-models/blob/v1.17.0/iCub/robots/iCubGazeboV2_5/model.urdf#L1815): 0.636185 Kg. 

Note that the actual value could differ from those due to differences between the real object and the CAD model.

traversaro, 2020-10-22:
@KhomutovEvgeniy  if this reply is ok for you, feel free to close the issue, thanks!

closed 2020-10-22 by KhomutovEvgeniy