robotology/icub-models #38: [Q/A] Query regarding Gazebo foot collision elements

prashanthr05, 2020-07-01:
Few SDF collision elements [here](https://github.com/robotology/icub-models/blob/371cc30d27f079d4407e209395ff6a691293abc1/iCub/robots/iCubGazeboV2_5/model.urdf#L2237), for the foot seems to be outside the parent tag `collision`.

``` xml
  <maxContacts>4</maxContacts>
  <kp>18000000</kp>
  <kd>100</kd>
  <maxVel>100</maxVel>
  <minDepth>0.0001</minDepth>
  <mu1>1</mu1>
  <mu2>1</mu2>
  <fdir1>0 0 0</fdir1>
```

Also, http://sdformat.org/spec?elem=collision&ver=1.6 specifies `max_contacts` tag. Moreover, the parameters `kp`, `kd`, `mu1`, etc fall as children of `friction`->`physics_engine` (eg physics_engine `ode`) tags.

There seems to be a mismatch between the SDF format and the gazebo reference tags written in the robot URDF file. Is this because we use the SDF tags within a URDF format?

prashanthr05, 2020-07-01:
cc @nunoguedelha @traversaro @GiulioRomualdi

GiulioRomualdi, 2020-07-01:
I think that also @diegoferigo may know the answer

diegoferigo, 2020-07-01:
Here is the answer: http://gazebosim.org/tutorials?tut=ros_urdf#%3Cgazebo%3EElementsForLinks

prashanthr05, 2020-07-01:
Great, Thanks!

closed 2020-07-01 by prashanthr05