robotology/icub-models #39: Problems while loading models in Gazebo 11

pattacini, 2020-07-05:
Hi,

I've recently started off playing with Gazebo 11 in a Docker Image and I've noticed that there are models provided in this repo that will crumble down to the ground as soon as loaded within the environment.

Here are a couple of examples:

| |
| :---: |
| ![][1] |
| Model working OK |
| ![][2] |
| Model NOT working |

As @traversaro found out, it might be due to https://github.com/osrf/gazebo/issues/2728.

[1]: https://user-images.githubusercontent.com/3738070/86529513-79ca2080-beb1-11ea-98ec-180d99453c96.gif
[2]: https://user-images.githubusercontent.com/3738070/86529519-80f12e80-beb1-11ea-9ae5-924a32b3ce6a.gif

diegoferigo, 2020-07-06:
I just checked and also my setup is affected by this problem. I confirm that the `(no hands)` models works fine.

traversaro, 2020-07-06:
The reason why the `(no hands)` model is working fine is that it is not a URDF file included in an SDF, so it is not affected by https://github.com/osrf/gazebo/issues/2728 . On the other hand, the feet_fixed and fixed  models are indeed URDF models included in an SDF (see https://github.com/robotology/icub-models/blob/master/fixed_model.sdf.in), so they are indeed affected by https://github.com/osrf/gazebo/issues/2728 . As the SDF files are already automatically generated, it should be sufficient to just generate them with version 1.7 if Gazebo 11 is found  by CMake.

pattacini, 2020-07-06:
To sum up, we're required to include the patch you recommended, @traversaro?

traversaro, 2020-07-06:
> 
> 
> To sum up, we're required to include the patch you recommended [here](https://github.com/osrf/gazebo/issues/2728#issuecomment-631474681), @traversaro?

More and less, basically that patch works fine if the `sdf`  files are already in the source repo, while in the case of this repo we need to modify the sdf files that we already automatically generated.

closed 2020-07-07 by traversaro