robotology/icub-models #41: Add the iTeen model

S-Dafarra, 2020-07-20:
In order to start performing experiments with the iTeen robot, it is fundamental to have its urdf and the possibility of using it in Gazebo.

With this issue, I would like to track the generation and the issues arising when trying to use this model for the first time.

cc @pattacini, @traversaro, @fiorisi

pattacini, 2020-07-23:
@S-Dafarra a quick update on this.

Our responsible person for generating the URDF is @salvi-mattia 👨🏻‍💼
We've organized an internal course to transfer the knowledge of exporting models that will take place next week.

More in general, we're set to proceed iteratively aiming at the following MVPs:
1. URDF of Torso + Head.
1. URDF of Torso + Head + Limbs.
1. URDF of The whole robot including fingers.

The exact number, as well as the content of MVPs, will be suitably detailed/discussed throughout the process.

Also, we will invite you periodically to attend our internal updates.

cc @DanielePucci @maggia80

S-Dafarra, 2020-07-23:
Sounds good, thanks @pattacini

pattacini, 2020-08-03:
Hi @S-Dafarra 

We've just invited you to `icub-tech-iit/cad-mechanics` where @salvi-mattia will be opening up a series of issues to address activities on the CAD side. You can follow the progress there.

We'll be using this issue instead to report on specific news regarding the URDF _per se_.

lornat75, 2020-08-04:
cc @xEnVrE

salvi-mattia, 2020-08-07:
hi everyone,
[these](https://github.com/icub-tech-iit/cad-mechanics/wiki/Creo-Mechanism-to-URDF) are the guidelines we will follow building up the URDF files foriCub3.
In the next weeks I'll work on the cad related part.
Following the handson and talking with @fiorisi we think in could be useful to generate different URDF related tu anatomical robot'a parts and assemble them at URDF level.
[ie different icubs share the same head, we often build partial robots, it could be possible to change the hands as they will be developed]
So we thought about this modules:
- torso, shoulders and upperarms
- head
- legs
- hands and forearms
we will need to define "universally accepted" interfaces between the modules

@S-Dafarra @FabioBergonti @fiorisi any opinions?

fiorisi, 2020-08-07:
I agree @salvi-mattia.
If there is a simple way ([xacro](http://wiki.ros.org/xacro)?) to reassemble the URDF modules is the best way to handle different robot versions and upgrades (e.g. new foreamr-hand system).

traversaro, 2020-08-07:
I think it is a great idea! I am not sure about what is the best way of composing the part URDF to generate a complete URDF, but we can sort that out.

salvi-mattia, 2020-08-12:
[Here](https://github.com/icub-tech-iit/cad-mechanics-public/wiki/Creo-mechanism-to-URDF) you can find the last version of the procedure.

It is public and it will be the official version for this procedure, we will keep this one up to date

DanielePucci, 2020-08-12:
CC @FabioBergonti @vpunithreddy @gabrielenava

salvi-mattia, 2020-08-13:
looking @ iCub2 simulation model, the head just include the three DOF in the neck, do we wanto to have also eyes' movementes?

DanielePucci, 2020-08-13:
Hi @salvi-mattia, my take is that in the long run for sure, but I would say that they are not necessary for the first simulations of whole-body dynamic control and locomotion. So, if the process of adding the eyes' movements slows down considerably (e.g. days) the release date of the model, I would say to avoid adding them unless the DOFs are necessary in the short term to other research lines

CC
@lornat75 @ChiaraBartolozzi

pattacini, 2020-08-13:
The eyes are traditionally a dull pain to deal with in Gazebo (@xEnVrE knows it quite well), so we definitely need to address the automatic generation.

That being said, I agree that for the first MVP we could skip them and proceed faster to then refine over if those small mechanical details regarding eyes don't turn out to be a walk in the park.

salvi-mattia, 2020-08-13:
designing robots is traditionally way better than a walk in the park.

salvi-mattia, 2020-08-14:
I have fixed some fail in the iCub3 model,
at the moment it weighs 55 kg [in the CAD]
while the actual robot is 51 kilos
this error is not that easy to fix because it's not easy to spot where the problem is really located.

do we think we could go ahead with this mass properties?

@S-Dafarra @fiorisi @DanielePucci

pattacini, 2020-08-14:
Nicely done @salvi-mattia 👍🏻 
I remember those old times when the weight discrepancy was used to amount to tens of kilos 😄 

My hunch is that we could proceed as it is now and then once we'll have the first overall exportable model we can backtrace and bisect the CAD in order to narrow down the problem. This means that we will very likely need eventually to "bisect" the real robot in order to measure the weights of the single parts as well.

fiorisi, 2020-08-14:
@salvi-mattia for the iCub2.5 model we had to ["manually" adjust the weight](https://github.com/robotology/icub-model-generator/blob/9f6dbe7ab98c4a4762317e2c9aacb298ffe39c53/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in#L932-L939) of the robot by reducing the mass of the sub assemblies in the second part of the URDF generation process (from the SimMechanics model to the URDF). 

See [this Excel table](https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub2_5/ConversionProcess.xlsx) to have an idea.

For this it would be nice to have an idea about the weight error for the main sub-assemblies (e.g. lower body, upper body and head).

[In this issue](https://github.com/robotology/icub-tech-support/issues/98), some of the robot sub-assemblies were weighted. It should be already possible to check the error.

fiorisi, 2020-08-14:
One of the [prerequisites](https://github.com/icub-tech-iit/cad-mechanics-public/wiki/Creo-mechanism-to-URDF#preliminary-operations) is to have a clear list of joint and link names. Something like [this](http://wiki.icub.org/wiki/ICub_Model_naming_conventions) (for iCub2). 

Otherwise, later on we will have to rename many files and change the associated script files.

salvi-mattia, 2020-08-14:
@S-Dafarra if you do have a list like that to provide it could be useful.
otherwise I'm naming the link following icub2 style and we will compose the table at the end [if smthg will be terribly wrong we con always rename some stuff]

DanielePucci, 2020-08-15:
@salvi-mattia 
> at the moment it weighs 55 kg [in the CAD]
while the actual robot is 51 kilos
this error is not that easy to fix because it's not easy to spot where the problem is really located.
>
> do we think we could go ahead with this mass properties?

I would love to weigh 4 kilos less than what my CAD model say 😄 With that said, I agree with @fiorisi  on

> [In this issue](https://github.com/robotology/icub-tech-support/issues/98), some of the robot sub-assemblies were weighted. It should be already possible to check the error.

So, if this process does not take too long, we would love to have already an idea where the main weight mismatch (CAD vs Reality) is. Consider that we could always unmount the robot (e.g. split it in two) and weigh it if we are in trouble (CC @Fabrizio69 @maggia80).

Unfortunately, back in time, *accurate weight* in the CAD was one of the important features to have in order to  achieve the [first balancing results](https://www.youtube.com/watch?v=jaTEbCsFp_M), and the mismatch that @pattacini [mentioned](https://github.com/robotology/icub-models/issues/41#issuecomment-674090060) was among the things that we had to correct. 

So, having an idea on where to correct the weight would later help the control algorithms, which make use of the CAD information extensively for (among other things) [center of pressure](https://en.wikipedia.org/wiki/Center_of_pressure_(terrestrial_locomotion)) and [center of mass](https://en.wikipedia.org/wiki/Center_of_mass) computations.

salvi-mattia, 2020-08-17:
hi all,
iCub3 model has a plenty of simp rep whose reliability we arn not sure.
I started creating a new bunch called Sim_Icub3_[link name]
es. Sim_Icub3_Chest
Sim_Icub3_R Elbow
Sim_Icub3_Shoulder_1
Sim_Icub3_Shoulder_2
Sim_Icub3_Upperarm

they still miss datums [axis, frames or planes] for further assembly

@FabioBergonti @fiorisi @Mick3Lozzo 
if you could please give a fast look at them we can check if we all understand the same concepts in creating sim reps

It could be time saving to spot systematic errors in the early part of the work.

@S-Dafarra i dont know if you have acces to the cad or someone can show you what we are talking about.

pattacini, 2020-08-17:
Yep, @S-Dafarra can read the CAD drawings.

salvi-mattia, 2020-08-17:
kudos 4 @FabioBergonti that make me notice that I didn't include in the "chest" part the small part of the neck that "doesn't move" but it's fixed with the sholuder.
that is true and that is because the connection between the chest and the head is not in a joint but "in the middle" of a link
to fix thi minor problem I suggest to use a "fixed" joint between these two parts [as it happend for the FTsensor that is a "fixed joint" in the middle of a link]

do you think it could be a good way to proceed?

@fiorisi @S-Dafarra

S-Dafarra, 2020-08-17:
> @S-Dafarra if you do have a list like that to provide it could be useful.
> otherwise I'm naming the link following icub2 style and we will compose the table at the end [if smthg will be terribly wrong we con always rename some stuff]

I don't have a list at the moment. When writing code, we tried to be as generic as possible. Clearly, having a list of joints similar to iCub2 would simplify the porting process. One thing that we usually assume is that there is a frame on the robot sole having the z axis pointing upward and the x axis pointing forward. These frames are called ``l_sole`` and ``r_sole`` and are kind of omnipresent in our codes. Another thing pretty common is the ``root_link``. Apart from these few features, we are pretty open for the naming.

> @S-Dafarra I don't know if you have access to the cad or someone can show you what we are talking about.

I do have access to the repo, but I am not sure if I can provide any useful comment on this :grin: For sure I appreciate any effort in making the process of generating URDF models more reliably. 

> do you think it could be a good way to proceed?

I don't think it is a problem having an additional fixed joint. In view of having the URDF composable, my two cents would be to include the blue link to the head subassembly if this part can be changed according to the installed head, to the chest otherwise.

salvi-mattia, 2020-08-17:
I think Ihave integrated different modifications suggested by @FabioBergonti 

as we said [here](https://github.com/robotology/icub-models/issues/41#issuecomment-670546542)

I think we have now the simp reps to complete the first module: **torso, shoulders and upperarms**

@fiorisi if you could please give a fast check we will procede adding datums and making shrinkwrap
so we can start validating the process in the next [non-CAD] steps

[image]

fiorisi, 2020-08-18:
Great @salvi-mattia! 🚀 
I had a quick look at the simplified rep and to me they look fine. Just a few comments:
- `iCub3 model has plenty of simp rep whose reliability we are not sure about.` I would delete all the unnecessary `Sim_` SimpRep.
- `I started creating a new bunch called Sim_Icub3_[link name]`. For the new SimpReps let's follow the [naming convention](https://github.com/icub-tech-iit/cad-mechanics-public/wiki/Creo-mechanism-to-URDF#how-to-define-the-simplified-representations).
- `I suggest to use a "fixed" joint between these two parts` the lower neck is part of the chest, I wouldn't divide it from the neck.

fiorisi, 2020-08-18:
> I suggest to use a "fixed" joint between these two parts the lower neck is part of the chest, I wouldn't divide it from the neck.

After a T2T meeting with @salvi-mattia we agreed that this division is the interface between the head assembly URDF and upperbody assembly URDF.

salvi-mattia, 2020-08-26:
together with @fiorisi we are facing various problems in naming conventions, simp rep separations, joint naming...
we are trying to overcome them not only for the contingency but to create a more reliable procedure [hoping it will help to be faster in the future]

@fiorisi @S-Dafarra fix is glad to invite you to our review in order to take stock of the situation.
it will take few minutes [15 I guess] starting from 2pm next friday [28th of august 2020]

S-Dafarra, 2020-08-26:
> @fiorisi @S-Dafarra fix is glad to invite you to our review in order to take stock of the situation.
> it will take few minutes [15 I guess] starting from 2pm next friday [28th of august 2020]

Thanks for the invitation, but unfortunately I will be busy at that time.

pattacini, 2020-08-26:
@S-Dafarra our review meeting will start at 2:00 pm and is supposed to end at 4:30 pm, so you happen to get a free 15 mins slot in this range, feel free to get back to us.

fiorisi, 2020-08-27:
> @fiorisi @S-Dafarra fix is glad to invite you to our review in order to take stock of the situation.
it will take few minutes [15 I guess] starting from 2pm next friday [28th of august 2020]

Thanks @salvi-mattia, `dev` has also the review meeting at the same time.

lornat75, 2020-08-27:
good job! I agree that having the eyes in the model is not urgent for the balancing tasks, however, it would be nice to have them.

S-Dafarra, 2020-08-28:
> @S-Dafarra our review meeting will start at 2:00 pm and is supposed to end at 4:30 pm, so if you happen to have a free 15 mins slot in this range, feel free to get back to us.

Actually, I realized I confused this Friday with the next one. I will definitely pass by today.

pattacini, 2020-08-28:
So we're gonna start off w/ iCub3 URDF at 14:00 if it's ok.
cc @S-Dafarra @salvi-mattia 

@xEnVrE if you want to attend too you're more than welcome, although we're not going to cover details like fingers and eye bulbs at this stage that you're more interested in.

xEnVrE, 2020-08-28:
@pattacini thanks for the invitation. I have two meetings in that slot but if the second ends earlier I will join for sure!

salvi-mattia, 2020-08-28:
We have completed the mechanism for the first assembly.
as we said [here](https://github.com/robotology/icub-models/issues/41#issuecomment-670546542) it represents torso, shoulders and upperarms.

@Nicogene is going to perform folloeing steps in the instructions pipeline and giving us feedback about his work and the problems he will face.

@fiorisi and @FabioBergonti have given a big help in naming links and joint and doublechecked my work step by step

@fiorisi is glad to help @Nicogene and whoevere else will work on this activity for the mechanical side.

if you update cad mechanics repository, you can find the mechanism assembly at this location

cad-mechanics\projects\simulation_model\icub3

opening it you can finally push the magic matlab button

Nicogene, 2020-08-31:
# First iteration of urdf generation

Today I opened the asm created by @salvi-mattia for the upperbody and I pressed the magic button.
Despite a warning of Creo I obtained a list of stl and an xml file that I used for obtaining a first version of the urdf.

I simply run:

```
simmechanics_to_urdf SIM_ICUB3_UPPERBODY.xml --yaml icub3-upperbody-options.yaml --outputfile test.urdf
```

Where in the yaml I simply specified the scale factor of 0.001(Creo uses mm, gazebo m)

And here is the result, the upperbody sunbathing in gazebo 🏖️ 

[image]

The next steps will include for sure the definition of joint limits through the csv file and maybe the renaming the link?

Nicogene, 2020-08-31:
FYI the work is carried on this branch https://github.com/robotology/icub-model-generator/tree/feat/addiCub3Upperbody

salvi-mattia, 2020-09-14:
As we decided today during fix's sprint planning I'm going to produce simp rep, shrinkwrap and mechanism for the legs, later on we will check if it's better to assemble them to the torso at CAD or URDF level.

In the mean while we will check how the ftsensor in the upperarms and its fixed joints works in this procedure

Nicogene, 2020-09-15:
After a meeting with @salvi-mattia @FabioBergonti and @traversaro, it emerged the necessity of having the link reference frames in the CAD before the conversion to the urdf.
If not defined it is generated with the z-axis passing through the joint axis, with y-x direction arbitrary.
Moreover, different generations can give different axis configurations.

For these reasons before proceeding with the work on the legs, we would first define those frames [following this convention](https://www.ros.org/reps/rep-0103.html#axis-orientation).
With this naming convention: `SCSYS_<child_link_name>`.

Is it ok for you?

cc @pattacini @DanielePucci @S-Dafarra @maggia80

traversaro, 2020-09-15:
Note that in the future we may need to also export a frame aligned with the traditional iCub http://wiki.icub.org/wiki/ICubForwardKinematics , but that is something that we can easily do later just for the root, while for all the other links I think it it is beneficial to use the REP 103 convention, as we already did when we clearly defined the frames for the legs in https://github.com/robotology/icub-model-generator/commit/97827f894a2572b10697a5a33639b530690151b0 .

traversaro, 2020-09-16:
> For these reasons before proceeding with the work on the legs, we would first define those frames [following this convention](https://www.ros.org/reps/rep-0103.html#axis-orientation).
> With this naming convention: `SCSYS_<child_link_name>`.
> 
> Is it ok for you?

Note that this should avoid all the problems that we had in https://github.com/robotology/icub-models/issues/20 .

salvi-mattia, 2020-09-18:
this is icub3 upper body seen from top
[image]
if I correctly interpret 
[image]
X axis will be placed like the yellow one and not like the blue

it means that we will have a lot of joint axis coincident with no frame axis.

is it correct?
is it the way we want to proced?

@S-Dafarra @pattacini @DanielePucci @traversaro @fiorisi @FabioBergonti

pattacini, 2020-09-18:
@salvi-mattia it sounds a bit weird.

traversaro, 2020-09-18:
Both options have their advantages, some days ago with @salvi-mattia I defended the "yellow arrow"-option but if the plan in the long term is to consider each limb as a standalone robot, probably the option that makes more sense even in accordance to rep-103 is the "blue arrow", sorry if I changed my mind.

S-Dafarra, 2020-09-18:
To be honest, I am not able to find particular advantages in having the yellow solution, so I would prefer the blue one and have at least one axis aligned with the axis of rotation.

Am I missing something?

traversaro, 2020-09-18:
In general, having parallel frames in at least one pose of the robot drastically simplifies the debugging of vectors expressed in different link frame, as it is not necessary to rotate them, or for position vectors you just need to sum them to the origin of the frame, and that is the reason why most URDF assign the link orientations to be parallel to each other, as described in REP-103 .

S-Dafarra, 2020-09-18:
> In general, having parallel frames in at least one pose of the robot drastically simplifies the debugging of vectors expressed in different link frame, as it is not necessary to rotate them, or for [...]

Ah I see, thanks. Anyways, if the robot moves from the zero position, anyway those frames will not be aligned anymore right, especially on the yellow case?

salvi-mattia, 2020-09-18:
talking with @traversaro it seems to be a good solution to have a definition with no uncertainty dependig on robot position.

our proposal [shared with @FabioBergonti ] could be:

Z axis on the joint axis
X axis on the link main axis [pointing next joint]
Y axis according to dextrous convention

If no one spots problem in this kind of convention we can proceed and see how it will work

traversaro, 2020-09-18:
Just to clarify instead the advantage of the "all links in the same limb" strategy:
* It is coherent we the strategy we already adopted with the

traversaro, 2020-09-18:
> I can list a few as soon if I have time

Humanoid models that follow the convention "all frame have the same orientation in a given configuration): 
* Boston Dynamics' Atlas: https://github.com/osrf/drcsim/blob/master/atlas_description/urdf/atlas.urdf
* NASA' Valkyrie: https://github.com/gkjohnson/nasa-urdf-robots/blob/master/val_description/model/robots/valkyrie_A.urdf
* SoftBank Robotics Nao: https://github.com/ros-naoqi/nao_robot/blob/master/nao_description/urdf/naoV50_generated_urdf/nao.urdf 
* PAL Robotics' REEM C and Talos: https://github.com/pal-robotics/talos_robot/blob/kinetic-devel/talos_description/urdf/arm/arm.urdf.xacro 
* ROBOTIS' OP3 : https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Common/blob/0780e54c2064a42089b0cf970d716ae44fec9233/op3_description/urdf/robotis_op3.structure.lleg.xacro
* Shadow Hand (not a humanoid but indeed a complex articulated model): https://github.com/shadow-robot/sr_common/blob/melodic-devel/sr_description/hand/xacro/finger/distal/distal.urdf.xacro 

The only humanoid URDF that is not using this convention is the iCub 2.5, that is using this convention for the legs but for the arms has arbitrary frame coming out from the Creo generation process (see https://github.com/robotology/icub-models/issues/20).

> >  In general, having parallel frames in at least one pose of the robot drastically simplifies the debugging of vectors expressed in different link frame, as it is not necessary to rotate them, or for position vectors you just need to sum them to the origin of the frame, and that is the reason why most URDF assign the link orientations to be parallel to each other, as described in REP-103 .
> 
> Ah I see, thanks. Anyways, if the robot moves from the zero position, those frames will not be aligned anymore right, especially on the yellow case?

Exactly.

pattacini, 2020-09-18:
Are we maybe discussing about 3 different conventions?
- the "yellow-arrow";
- the "iCub 2.5 legs";
- the one proposed in https://github.com/robotology/icub-models/issues/41#issuecomment-694780076.

That one @traversaro seems to support is not the so-called "yellow-arrow" but rather that one already implemented in the iCub 2.5 legs. On the other hand, @salvi-mattia, yours is more "traditional".

Did I get this correct?

traversaro, 2020-09-18:
> Did I get this correct?

Exactly. The main difference between "yellow-arrow" and "iCub 2.5 legs" is at which level you try to have link frame all with the same orientation. For  "yellow-arrow"  it is at the robot level (that given iCub shape leads to peculiar behavior) while "iCub 2.5 legs" is just to keep the same orientation for all the links in a given limb. 

>  X axis on the link main axis [pointing next joint]

Small note: with this convention you still have have a small ambiguity (the same that you have in DH) when assigning frames for a joint whose axis coincide (such as the ankle or hip in iCub).

kouroshD, 2020-09-18:
> Humanoid models that follow the convention "all frame have the same orientation in a given configuration):

I agree with @traversaro . Back in time, when I was using the different robot's urdf for the retargeting purposes, I had this problem with iCub2.5 urdf model. Instead, other robots were straight forward to do retargeting.

I do not agree with this comment https://github.com/robotology/icub-models/issues/41#issuecomment-694780076 , since the robot motor axis can be different, and we will end up with urdf model that every link has different frame orientations.  But instead, putting all the joint to N-pose or T-pose (or zero values for all the motor joint angles) we will have a much cleaner urdf of the robot.

traversaro, 2020-09-18:
Hi @S-Dafarra @kouroshD , if you could discuss internally with other DIC people and reach a consensus on this it would be great, otherwise the risk is that the discussion become a bit too dispersive.

S-Dafarra, 2020-09-21:
> Hi @S-Dafarra @kouroshD , if you could discuss internally with other DIC people and reach a consensus on this it would be great, otherwise the risk is that the discussion become a bit too dispersive.

Sorry to show up just now. We discussed internally and we agreed on the following. The convention we prefer is defined as follows:
- When all the joints are in the zero position, all the links belonging to a limb share the same orientation. E.g., all the links of the torso would be aligned with frame A, all the links of the left arm with frame B, the right arm with frame C,....
- These "common" frames (A, B, C, ...) have the z-axis pointing upward, the x-axis pointing forward (clearly, upward and forward depend on the limb)

Given the particular design of the shoulders, in the zero position, a frame on an arm link is slightly rotated around the z-axis compared to a frame in the torso.

This convention is basically the one defined as "iCub 2.5 legs" in https://github.com/robotology/icub-models/issues/41#issuecomment-694884201, which corresponds to the blue arrow of https://github.com/robotology/icub-models/issues/41#issuecomment-694738366.

salvi-mattia, 2020-09-23:
after a meeting with @S-Dafarra @FabioBergonti and @pattacini 
we agreed on the following:
- every link has a solidal frame with 
- X axis pointing forward
- Z axis pointing up
- Y axis to have a dextrous frame
- one of the axis has to be coincident with the joint axis

when it's not applicable (in plaine words, we may derive a better formalism later):
- one axis is coincident with joint axis
- Z axis is the one pointing upwards at the most
- X axis pointing forward at the most
- Y axis to have a dextrous frame

salvi-mattia, 2020-09-24:
[image]

salvi-mattia, 2020-09-30:
I added the frames in the linke and reassembled the mechanism.
Everything is pushed in cad-mechanics
@Nicogene u can now make your magic-button-stuff

after that it could be useful to check if we are getting what we want, so that we can proceed with the legs in the correct way

Nicogene, 2020-10-01:
@S-Dafarra in [icub3-upperbody.zip](https://github.com/robotology/icub-models/files/5311133/icub3-upperbody.zip) you should find all you need for opening the urdf in gazebo, can you check if the frames are following the right convention?

S-Dafarra, 2020-10-01:
> @salvi-mattia added and checked the link frames but 4 eyes are better than two 
> 
> @S-Dafarra could you please double-check if the frames are following the right convention?
> 
> In [icub3-upperbody.zip](https://github.com/robotology/icub-models/files/5311133/icub3-upperbody.zip) you should find all you need for opening the urdf in gazebo.
> Please let me know if you experience any issue.

Hi @Nicogene, I just tried, but I am not able to load the model. I get the following error
```
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_root_link_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_torso_1_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_torso_2_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_chest_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_l_shoulder_1_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_l_shoulder_2_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_l_shoulder_3_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_l_upperarm_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_r_shoulder_1_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_r_shoulder_2_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_r_shoulder_3_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [sim_icub3_r_upperarm_prt.stl]
[Err] [Visual.cc:2951] No mesh specified
```
I just specified the ``GAZEBO_MODEL_PATH``, should I add something else?

[33 more replies]

closed 2020-11-12 by traversaro