robotology/icub-models #46: Automatically generate the **_visuomanip  models from existing models

traversaro, 2020-07-27:
Similar to what is done for all others model stored in this repo, the original source of information (hands model, eyes models) should be stored in some other location, and then the complete models (such as `iCubGazeboV2_5_visuomanip`) should be generate automatically, to ensure that modifications in new version of iCub can be propagated automatically. 

In this issue in particular we track the modifications done in the  `iCubGazeboV2_5_visuomanip` model, to make sure that they will be eventually all ported to an automatic procedure. 

Know modifications: 
* [ ] A model with hands and eyes has been added in https://github.com/robotology/icub-models/pull/42 . Hands and eyes models are not derived from CAD models, but this should not stop us from having an automatic procedure: the "manually" prepared hands and eyes model should prepare a standalone models, that then are combined with the rest of the model coming from the CAD in an automatic way 
* [ ] The `head_imu_acc_1x1` sensor tag has been removed from a `<gazebo>` tag in https://github.com/robotology/icub-models/pull/45, but the `accelerometer` tags were never supposed to be there in the first place.

prashanthr05, 2021-03-23:
> The head_imu_acc_1x1 sensor tag has been removed from a <gazebo> tag in #45, but the accelerometer tags were never supposed to be there in the first place.

If I am not wrong, they might have been added for some iDynTree specific computations.

traversaro, 2021-03-23:
> 
> 
> > The head_imu_acc_1x1 sensor tag has been removed from a  tag in #45, but the accelerometer tags were never supposed to be there in the first place.
> 
> If I am not wrong, they might have been added for some iDynTree specific computations. But wrongly added within the `<gazebo>` tags?

Yes, it is probably something related to that.