robotology/icub-models #49: Add the waist-imu in iCubGazeboV2_5

GiulioRomualdi, 2020-09-28:
Every time we have to use the waist IMU in the simulation we need to switch to a custom branch developed by @prashanthr05. Please check [here](https://github.com/prashanthr05/icub-models/blob/eedf2ed2b6ffe0e6523a30db14821a45d8f6c29c/iCub/robots/iCubGazeboV2_5/model.urdf).

 It would be nice if the IMU is embedded by default in the simulation model.

traversaro, 2020-09-28:
Hi @GiulioRomualdi ! 

As documented in https://icub-tech-iit.github.io/documentation/icub_versions/, the high performance Xsens MTi mounted in the waist of the robot is a feature introduced in iCub 2.7, so rather then adding it to the iCubGazeboV2_5 model, probably it is better to introduce a iCubGazeboV2_7 model that contains this additional frame (and ideally, also the updated nominal inertial parameters for the waist with this additional IMU). 

There are two ways of doing so: 
1. Ensure that a new Creo mechanism model is generated that matches the iCub 2.7 configuration in https://github.com/icub-tech-iit/cad-mechanics-public/tree/master/projects/simulation_model/icub1_icub2, and from that one generate a SimMechanics model to add to this repo. This will probably take time and the availability of a Mech Engineering to update the model.
2. Modify the generation scripts in  https://github.com/robotology/icub-model-generator to create a iCubGazeboV2_7 model starting from the iCubGazeboV2_5 model, and just adding the `root_link_imu_frame` frame, for example by inserting the frame location directly via a `XMLBlobs`, see for example https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_simmechanics_options.yaml.in#L944 .

The best long term solution is the 1. , but 2. is probably the best solution in the short term. 

Just to clarify, the custom branch in https://github.com/prashanthr05/icub-models/blob/eedf2ed2b6ffe0e6523a30db14821a45d8f6c29c/iCub/robots/iCubGazeboV2_5/model.urdf contains a lot of changes, but in this issue let's discuss about the waist IMU frame.

Nicogene, 2020-11-09:
The related PR has been merged, closing

closed 2020-11-09 by Nicogene