robotology/icub-models #51: Add sanity checks for iCub3 models

traversaro, 2020-10-17:
To avoid strange regressions, we have some sanity checks for the iCub 2.* models in https://github.com/robotology/icub-model-generator/blob/master/tests/CMakeLists.txt#L25 . It probably could make sense to have similar checks for iCub 3 models, to avoid regressions such as the one described in https://github.com/robotology/icub-models/issues/41#issuecomment-710791312 .

Nicogene, 2020-10-21:
This is partially fixed by https://github.com/robotology/icub-model-generator/pull/149.

Using the sanity checks emerged the difference between icub2 and icub3 for what concerns the orientation of ft sensors

traversaro, 2020-10-22:
Fixed by https://github.com/robotology/icub-model-generator/pull/149 . I guess that if the need for new tests arise, they can be added to the test suite, no need for a separate issue tracking this.

closed 2020-10-22 by traversaro