robotology/icub-models #58: Error while loading iCubGazeboV3_fixed

GiulioRomualdi, 2020-10-23:
When I tried to load `iCubGazeboV3_fixed`  an error is thrown by gazebo 11. 
Consequentially the robot is not fixed in the world

Here the error:
```
[Err] [SdfFrameSemantics.cc:91] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [fixed_base] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [SdfFrameSemantics.cc:91] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [fixed_base] does not have FrameType LINK OR MODEL when starting from vertex with name [fixed_base].
[Err] [SdfFrameSemantics.cc:91] Error Code 26 Msg: PoseRelativeToGraph error, Vertex with name [fixed_base] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[Err] [SdfFrameSemantics.cc:91] Error Code 26 Msg: PoseRelativeToGraph frame with name [fixed_base] is disconnected; its source vertex has name [fixed_base], but its source name should be __model__.
[Err] [Model.cc:99] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [fixed_base] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[Err] [Model.cc:99] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [fixed_base] does not have FrameType LINK OR MODEL when starting from vertex with name [fixed_base].
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph error, Vertex with name [fixed_base] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph frame with name [fixed_base] is disconnected; its source vertex has name [fixed_base], but its source name should be __model__.
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/left_leg/analog:o/rpc:i active at tcp://192.168.1.3:10004/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/left_leg/analog:o active at tcp://192.168.1.3:10005/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/left_foot_front/analog:o/rpc:i active at tcp://192.168.1.3:10006/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/left_foot_front/analog:o active at tcp://192.168.1.3:10007/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/left_foot_rear/analog:o/rpc:i active at tcp://192.168.1.3:10008/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/left_foot_rear/analog:o active at tcp://192.168.1.3:10009/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/right_leg/analog:o/rpc:i active at tcp://192.168.1.3:10010/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/right_leg/analog:o active at tcp://192.168.1.3:10011/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/right_foot_front/analog:o/rpc:i active at tcp://192.168.1.3:10012/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/right_foot_front/analog:o active at tcp://192.168.1.3:10013/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/right_foot_rear/analog:o/rpc:i active at tcp://192.168.1.3:10014/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/right_foot_rear/analog:o active at tcp://192.168.1.3:10015/
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubSim/left_arm/analog:o/rpc:i active at tcp://192.168.1.3:10016/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[INFO]created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
yarp: Port /icubSim/left_arm/analog:o active at tcp://192.168.1.3:10017/
[Err] [Joint.cc:297] EXCEPTION: Couldn't Find Child Link[iCub::base_link]

[Err] [Model.cc:247] LoadJoint Failed
[INFO]No ROS group found in config file ... skipping ROS initialization. 
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
[Wrn] [msgs.cc:1841] Conversion of sensor type[force_torque] not supported.
yarp: Port /icubSim/right_arm/analog:o/rpc:i active at tcp://192.168.1.3:10018/
[INFO]AnalogServer : no ROS initialization required 
[INFO]created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[... 158 lines cut ...]
[DEBUG]found:  iCub::r_shoulder_pitch r_shoulder_pitch 
[DEBUG]found:  iCub::r_shoulder_roll r_shoulder_roll 
[DEBUG]found:  iCub::r_shoulder_yaw r_shoulder_yaw 
[DEBUG]found:  iCub::r_elbow r_elbow 
[DEBUG]found:  iCub::r_wrist_prosup r_wrist_prosup 
[DEBUG]found:  iCub::r_wrist_pitch r_wrist_pitch 
[DEBUG]found:  iCub::r_wrist_yaw r_wrist_yaw 
[DEBUG]done 
[WARNING]No positionToleranceLinear value found in ini file, default one will be used! 
[WARNING]No positionToleranceRevolute value found in ini file, default one will be used! 
[WARNING]No max torques value found in ini file, default one will be used! 
[DEBUG]max_torques: [   2000.000000	 2000.000000	 2000.000000	 2000.000000	 2000.000000	 2000.000000	 2000.000000  ] 
[INFO]min_stiffness param found! 
[INFO]max_stiffness param found! 
[INFO]min_damping param found! 
[INFO]max_damping param found! 
[DEBUG]min_stiffness: [   0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000  ] 
[DEBUG]max_stiffness: [   1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000  ] 
[DEBUG]min_damping: [   0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000  ] 
[DEBUG]max_damping: [   100.000000	 100.000000	 100.000000	 100.000000	 100.000000	 100.000000	 100.000000  ] 
[WARNING]VELOCITY_CONTROL: 'velocityControlImplementationType' param missing.  Using the default value of 'direct_velocity_pid', but the parameter will be compulsory gazebo-yarp-plugins 4.
[DEBUG]INITIAL CONFIGURATION IS:  -0.520000	 0.520000	 0.000000	 0.785000	 0.000000	 0.000000	 0.000000 
[DEBUG]Initializing Trajectory Generator with default values 
[INFO]created device <gazebo_controlboard>. See C++ class GazeboYarpControlBoardDriver for documentation.
```

GiulioRomualdi, 2020-10-23:
Possible related issue: https://github.com/PX4/sitl_gazebo/issues/536

GiulioRomualdi, 2020-10-23:
I've just noticed that the same error occurs also with `iCubGazeboV2_5_fixed`. Gazebo version `11.1.0`

traversaro, 2020-10-23:
It is a recent regression caused by a recent PR, we should fix it by changing base_link to root_link

traversaro, 2020-10-23:
In particular, it is a regression of https://github.com/robotology/icub-model-generator/pull/142 . The fix is simply to change https://github.com/robotology/icub-models/blob/master/fixed_model.sdf.in#L11 to from `iCub::base_link` to `iCub::root_link` .

GiulioRomualdi, 2020-10-23:
I'm opening a PR

PR #59 (2020-10-23): Fix the child link in the "fixed_base" joint for fixed_model.sdf.in
This commit is required because of robotology/icub-model-generator#142

The PR fixes #58
closed 2020-10-23 by GiulioRomualdi