robotology/icub-models #60: [iCubGazeboV3] Ankles roll goes in HF for a given value of ankle pitch

GiulioRomualdi, 2020-10-23:
If the ankle pitch of the left foot is greater than 13deg the ankle roll goes in HF

[image]

The same behavior happens also for the right foot.
[image]

 Notice that the joint axis of the two ankles seems to be not coherent (associate issue

traversaro, 2020-10-23:
The HF in the simulated robot is raised whenever the requested torque is higher then 2000 Nm, see https://github.com/robotology/gazebo-yarp-plugins/blob/52491d69365a32ad930f2b858b9f3a43e7596510/plugins/controlboard/src/ControlBoardDriver.cpp#L69 and https://github.com/robotology/gazebo-yarp-plugins/blob/52491d69365a32ad930f2b858b9f3a43e7596510/plugins/controlboard/src/ControlBoardDriver.cpp#L568 . What is happened there at the first glance seems related to some numerical error that leads the simulated system to diverge once controlled, i.e. a problem similar to https://github.com/robotology/icub-models/issues/53 . Does the problem disappears if you increase some inertias of some links of the foot?

GiulioRomualdi, 2020-10-23:
Hi @traversaro changing the `max_step_size` from `0,001` to `0.0001` fixes the problem

Nicogene, 2020-10-23:
Probably we have to fake inertias also here

closed 2020-10-29 by Nicogene

Nicogene, 2020-10-29:
@GiulioRomualdi it should be fine now, can you double-check?