robotology/icub-models #61: [iCubGazeboV3] right leg joints direction wrongly defined

GiulioRomualdi, 2020-10-23:
The direction of the right joint seems to be wrongly defined 

This is `r_hip_roll`
[image]

This is for the `ankle pitch`
[image]

Notice the ankle roll goes in HF for a specific value of `ankle-pitch` for this problem please checkhttps://github.com/robotology/icub-models/issues/60

@traversaro and @Nicogene 
Similar issue: #57

GiulioRomualdi, 2020-10-23:
Regarding `r_hip_roll` I'm pretty sure it's a bug so I can open a PR in the generator. Following robotology/icub-model-generator#150.

 For the ankles, I don't know if it's wanted

traversaro, 2020-10-23:
> For the ankles, I don't know if it's wanted

As we don't have any document with the iCub3 direction, I guess the gold reference is the real robot.

GiulioRomualdi, 2020-10-26:
https://github.com/robotology/icub-model-generator/pull/153 and https://github.com/robotology/icub-model-generator/pull/151 fix the problems

closed 2020-10-26 by GiulioRomualdi