robotology/icub-models #62: [iCubGazeboV3][iCubGenova09] head_imu_acc_1x1/SCSYS_HEAD_MTX_IMU missing

prashanthr05, 2020-10-23:
We usually use the head IMU as the primary IMU to run the wholeBodyDynamics device in https://github.com/robotology/whole-body-estimators.
Currently, we are trying to configure wholeBodyDynamics on iCubGazeboV3 and iCubGenova09 models. This requires the availability of the head IMU. 

iCub3 SimMechanics XML file doesn't contain a `SCSYS_HEAD_MTX_IMU`. Should we generate it from the CAD?

See https://github.com/robotology/icub-model-generator/blob/master/simmechanics/data/icub3/ICUB3_ALL_SIM_MODEL.xml

On checking further the sim-mechanics file, there seems to be 3 coordinate systems CS1, CS2, and CS3 in the head link as seen in https://github.com/robotology/icub-model-generator/blob/49a4e0a4d11a09895c207baba1b29c5395ecc662/simmechanics/data/icub3/ICUB3_ALL_SIM_MODEL.xml#L2816

CS2 seems to be the frame `SCSYS_HEAD`. Is one of CS1 and CS3 related to the IMU?

How should we proceed if we would like to have the head IMU in the URDF model?

cc @traversaro @S-Dafarra @GiulioRomualdi

traversaro, 2020-10-23:
I do not know if the head IMU was included in this version of the model, probably @Nicogene knows something more on this. 
To avoid being blocked on this @prashanthr05, something you can do is to add a fake IMU in the simulation that is oriented as the head link, and start using it (in simulation) until the actual frame is available.

prashanthr05, 2020-10-23:
> To avoid being blocked on this @prashanthr05, something you can do is to add a fake IMU in the simulation that is oriented as the head link, and start using it (in simulation) until the actual frame is available.

Indeed. Thank you!

Nicogene, 2020-10-23:
Hi @prashanthr05, the frames of the sensors(except the ft-sensors) have not be defined yet, they will be added in a second draft of the model(for example we will add also the frames of the realsense in the chest).

>iCub3 SimMechanics XML file doesn't contain a SCSYS_HEAD_MTX_IMU. Should we generate it from the CAD?

Yes, they have to be defined in the cad and then exported in the xml-urdf.

cc @salvi-mattia

prashanthr05, 2020-10-23:
@Nicogene Thank you for the information. I will proceed with the workaround of the fake IMU in simulation.

salvi-mattia, 2020-11-16:
hi all,
the RFE MASTER board has got two different sensor: one from bosh and one from Invsense
which are we going to use?
whic measurement frame do we need?

pattacini, 2020-11-16:
>which are we going to use?

I think we're using the Bosch one.
A double-check is required anyway.

Nicogene, 2020-11-16:
I confirm Bosh

traversaro, 2020-11-16:
Yes, Bosch is the correct one, I am afraid that Invsense is not exposed at the firmware level.

Nicogene, 2020-11-19:
This should be fixed by https://github.com/robotology/icub-models-generator/pull/172

As soon as the CI finished, the model updated should be available, with the head v2.7 and then with the imu

Nicogene, 2020-11-19:
cc @prashanthr05

Nicogene, 2020-11-24:
Can we close it? Is it the imu fine?

prashanthr05, 2020-11-24:
> Can we close it? Is it the imu fine?

The Head IMU frames seem to have been added in this commit https://github.com/robotology/icub-models/commit/676cb8d5e53b791133987d89340253d196de20b6.

We have not tested it yet, but the purpose of this issue is sorted. We can close this issue. 
We will open a new issue in case if there are problems with the added frames.

Thank you @Nicogene @salvi-mattia @pattacini !

Nicogene, 2020-11-24:
I tagged you here (https://github.com/robotology/icub-models/commit/676cb8d5e53b791133987d89340253d196de20b6#r44490435) what is it missing or wrong?

prashanthr05, 2020-11-24:
> I tagged you here ([676cb8d#r44490435](https://github.com/robotology/icub-models/commit/676cb8d5e53b791133987d89340253d196de20b6#r44490435)) what is it missing or wrong?

I commented there that the required frame has been added. Thanks for pointing to the specific lines. 

I will close this issue.

closed 2020-11-24 by prashanthr05