robotology/icub-models #65: Fix correct location of RFE IMU for iCub 2.6 models

traversaro, 2020-11-03:
To correctly run WBD for robots that have a 2.6 version of the head, we need to correctly export the iCub models with the correct location of the `imu_frame` frame. To avoid the need to re-generate a simmechanics model, we can do something similar to https://github.com/robotology/icub-model-generator/pull/139 in which a new iCubGazeboV2_7 model is added with just modification at the configuration/CMake level.

traversaro, 2020-11-03:
See https://github.com/robotology/icub-main/issues/570 .

traversaro, 2020-11-03:
Once this is done, we should properly generate (or simply install) for the following robots that have RFE head: 
* `iCubGenova07` 
* `iCubHongKong01` 
* `iCubGenova08`

Nicogene, 2020-11-04:
For changing the pose of the frame of the IMU without passing through the cad we needed to add an option in the `simmechanics-to-urdf` script:

https://github.com/robotology/simmechanics-to-urdf/pull/42

Nicogene, 2020-11-06:
PR opened for producing the icub V2_6 model https://github.com/robotology/icub-model-generator/pull/164

Nicogene, 2020-11-09:
This should be done

closed 2020-11-09 by Nicogene