robotology/icub-models #67: Warning when iCubGazeboV2_5 spawn in gazebo

GiulioRomualdi, 2020-11-11:
When I try to load the `iCubGazeboV2_5` model the following warning is displayed.  

```
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[iCub].
```

I don't know if it's correlated but now I'm not able to load anymore the [`basestate`](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/basestate) plugin. 

I'm currently using the `devel` branch of the repo. The same happens also in master. 

cc @traversaro and @prashanthr05

GiulioRomualdi, 2020-11-11:
Is this related to this PR? https://github.com/robotology/icub-models/pull/59

traversaro, 2020-11-11:
> I don't know if it's correlated but now I'm not able to load anymore the [`basestate`](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/basestate) plugin.

 `basestate` is not part of the models provided in this repo, I guess you have some sort of composite model, can you point to it? Thanks!

GiulioRomualdi, 2020-11-11:
> `basestate` is not part of the models provided in this repo, I guess you have some sort of composite model, can you point to it? Thanks!

Here you have: https://github.com/GiulioRomualdi/icub-gazebo/tree/feature/icub-base-est/icub_base_est

traversaro, 2020-11-11:
This warning was added (if I am not wrong) in https://github.com/osrf/sdformat/commit/2266e4c8ec3cfa2849ca23defd8347b534d0a5fb , so it should be present just for Gazebo 11/SDFormat 9.

GiulioRomualdi, 2020-11-11:
I don't know if may help but `iCubGazeboV2_5_visuomanip` seems working without the warning

traversaro, 2020-11-11:
In https://github.com/robotology/icub-models/pull/40 we bumped the SDF version of iCub Gazebo models to be 1.7, just to avoid problems similar to the one mentioned in that issue  I suggest to try to bump the SDF version in https://github.com/GiulioRomualdi/icub-gazebo/tree/feature/icub-base-est/icub_base_est to 1.7 .

GiulioRomualdi, 2020-11-11:
> In #40 we bumped the SDF version of iCub Gazebo models to be 1.7, just to avoid problems similar to the one mentioned in that issue I suggest to try to bump the SDF version in https://github.com/GiulioRomualdi/icub-gazebo/tree/feature/icub-base-est/icub_base_est to 1.7 .

Thanks @traversaro this fixes the [`sdf` model](https://github.com/GiulioRomualdi/icub-gazebo/tree/feature/icub-base-est/icub_base_est)

traversaro, 2020-11-11:
> Thanks @traversaro this fixes the [`sdf` model](https://github.com/GiulioRomualdi/icub-gazebo/tree/feature/icub-base-est/icub_base_est)

Then probably this is yet another corner case of different version of SDF nested one into the other, see https://github.com/osrf/gazebo/issues/2728 and https://github.com/osrf/gazebo/issues/2739 . At this point, if you do not have any reason for supporting Gazebo <= 10, I would go for requiring Gazebo 11 and just change all the SDF files to be 1.7 .