robotology/icub-models #70: [iCubGenova09] Possible problems on the generated model

GiulioRomualdi, 2020-11-23:
On Friday we tested the walking controller on iCubGenova09. Before running the controller I updated the model to the latest commit devel 676cb8d5e53b791133987d89340253d196de20b. We noticed a strange movement of the swing foot while walking (like one of the joint axis has a wrong direction) . 
So we decided to use the model stored in our fork and everything went fine.
In particular, we decided to run the controller with the iCubGazeboV3 model stored in https://github.com/GiulioRomualdi/icub-models/blob/9a1cc2d6adcab85c2f5f1c18fb9acb1b71122fda/iCub/robots/iCubGazeboV3/model.urdf

This is the result of a `git` between the two models 
```
git diff custom_model_iCubGazeboV3.urdf official_iCubGenova09.urdf
```
- `custom_model_iCubGazeboV3.urdf`: https://github.com/GiulioRomualdi/icub-models/blob/9a1cc2d6adcab85c2f5f1c18fb9acb1b71122fda/iCub/robots/iCubGazeboV3/model.urdf
- `official_iCubGenova09.urdf`: https://github.com/robotology/icub-models/blob/676cb8d5e53b791133987d89340253d196de20b6/iCub/robots/iCubGenova09/model.urdf

The differences are really a lot
 
```patch
diff --git a/custom_model_iCubGazeboV3.model b/official_iCubGenova09.model
index cfee6a6..bf71763 100644
--- a/custom_model_iCubGazeboV3.model
+++ b/official_iCubGenova09.model
@@ -21,392 +21,260 @@
       </geometry>
     </collision>
   </link>
-  <link name="r_hip_1">
+  <link name="torso_1">
     <inertial>
-      <origin xyz="-0.005357899999999999 -0.03359610000000001 -0.001346999999999987" rpy="0 0 -1.5707963267948966"/>
-      <mass value="2.48546"/>
-      <inertia ixx="0.00285629" ixy="3.9100879313042506e-05" ixz="-4.3233113444168614e-05" iyy="0.00230094" iyz="2.8105985012852513e-05" izz="0.00246343"/>
+      <origin xyz="-0.00752632 0.000684313 0.05308400000000002" rpy="0 0 -1.5707963267948966"/>
+      <mass value="1.81255"/>
+      <inertia ixx="0.00354354" ixy="5.48743e-06" ixz="8.16841e-06" iyy="0.00272482" iyz="-0.000326014" izz="0.00215923"/>
     </inertial>
     <visual>
-      <origin xyz="-0.021600000000000008 0.04054999999999993 0.313858" rpy="0 0 -1.5707963267948966"/>
+      <origin xyz="0 0 0.25335800000000014" rpy="0 0 -1.5707963267948966"/>
       <geometry>
-        <mesh filename="package://iCub/meshes/simmechanics/sim_icub3_r_hip_1_prt.stl" scale="0.001 0.001 0.001"/>
+        <mesh filename="package://iCub/meshes/simmechanics/sim_icub3_torso_1_prt.stl" scale="0.001 0.001 0.001"/>
       </geometry>
       <material name="black">
         <color rgba="0 0 0 1"/>
       </material>
     </visual>
     <collision>
-      <origin xyz="-0.021600000000000008 0.04054999999999993 0.313858" rpy="0 0 -1.5707963267948966"/>
+      <origin xyz="0 0 0.25335800000000014" rpy="0 0 -1.5707963267948966"/>
       <geometry>
-        <mesh filename="package://iCub/meshes/simmechanics/sim_icub3_r_hip_1_prt.stl" scale="0.001 0.001 0.001"/>
+        <mesh filename="package://iCub/meshes/simmechanics/sim_icub3_torso_1_prt.stl" scale="0.001 0.001 0.001"/>
       </geometry>
     </collision>
   </link>
-  <joint name="r_hip_pitch" type="revolute">
-    <origin xyz="0.0216 -0.040549999999999996 0" rpy="0 0 0"/>
-    <axis xyz="0.0 -1.0 0.0"/>
+  <joint name="torso_roll" type="revolute">
+    <origin xyz="0 0 0.0605" rpy="0 0 0"/>
+    <axis xyz="-1.0 0.0 0.0"/>
     <parent link="root_link"/>
-    <child link="r_hip_1"/>
-    <limit effort="50000" lower="-0.7853981633974483" upper="2.007128639793479" velocity="120.0"/>
+    <child link="torso_1"/>
+    <limit effort="50000" lower="-0.4014257279586958" upper="0.4014257279586958" velocity="120.0"/>
     <dynamics damping="0.06" friction="0.0"/>
   </joint>
-  <link name="r_hip_2">
+  <link name="torso_2">
     <inertial>
-      <origin xyz="-0.0073287000000000005 -0.00018599999999999173 -0.03922199999999998" rpy="0 0 -1.5707963267948966"/>
-      <mass value="1.8834"/>
[... 1753 lines cut ...]
-    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.0"/>
+    <origin rpy="-3.141592653589793 -4.20000162559822e-15 -1.570796326794897" xyz="0.0034999999997785457 0.0054999999863750895 0.1013340000004441"/>
+  </sensor>
+  <gazebo reference="chest">
+    <sensor name="realsense_chest_rgb" type="camera">
+      <always_on>1</always_on>
+      <update_rate>30</update_rate>
+      <pose>0.0894115 0.0175 0.09929950000000001 -1.5707963267948968 -0.0 -1.5707963267948968</pose>
+      <plugin name="iCub_yarp_gazebo_plugin_camera" filename="libgazebo_yarp_camera.so">
+    <yarpConfigurationFile>model://iCub/conf_icub3/gazebo_icub_rgb_camera.ini</yarpConfigurationFile>
+</plugin>
+    </sensor>
+  </gazebo>
+  <sensor name="realsense_chest_rgb" type="camera">
+    <parent link="chest"/>
+    <origin rpy="-1.5707963267948968 -0.0 -1.5707963267948968" xyz="0.0894115 0.0175 0.09929950000000001"/>
   </sensor>
   <gazebo>
-    <plugin name="controlboard_wholebody_ros_nohands" filename="libgazebo_yarp_controlboard.so">
-        <yarpConfigurationFile>model://iCub/conf_icub3/gazebo_icub_wholebody_ros_nohands.ini</yarpConfigurationFile>
-    </plugin>
+    <pose>0.0 0.0 0.8 0.0 0.0 0.0</pose>
   </gazebo>
 </robot>
```

pattacini, 2020-11-23:
cc @Nicogene

traversaro, 2020-11-23:
A lof of difference are due to the fact that the order of the links in the two models are different.

traversaro, 2020-11-23:
> 
> 
> A lof of difference are due to the fact that the order of the links in the two models are different.

Ideally if we had a script to "normalize" a URDF, probably this would simplify checking the diff. We could also try XML-specific diff tools, that just look at the difference in matching elements, see https://extendsclass.com/xml-diff.html .

GiulioRomualdi, 2020-11-23:
@traversaro suggested that the problem has been caused by the fact that the new creo mechanism has been released in order to have iCub2.7 head on iCub3. Consequentially the `reverseRotationAxis` becomes outdated (e.g.    https://github.com/robotology/icub-models-generator/blob/e8e255a822a41d32c94aec30b5d3b1eef021af3d/simmechanics/data/icub3/ICUB_3_rightleg_options.yaml#L150-L154 )

In theory, we should recheck all the joints direction of the model with the robot. We should reenable the `icub-model-test` for iCub3

GiulioRomualdi, 2020-11-30:
I've just checked with @S-Dafarra and @prashanthr05 the axis direction by using `iCubGenova09`. The following joint axes are reversed:
- `l_shoulder_yaw`
- `l_wrist_prosup`
- `l_wrist_pitch`
- `l_wrist_yaw`

traversaro, 2020-11-30:
cc @Nicogene

GiulioRomualdi, 2020-11-30:
I can do that

pattacini, 2020-11-30:
>I can do that

Thanks @GiulioRomualdi 👍🏻

closed 2020-11-30 by traversaro