robotology/icub-models #71: [iCubGenova09/iCubGazeboV3] Check hip ft sensors

Nicogene, 2020-11-24:
After the fix of ft sensors representation(https://github.com/robotology/icub-models-generator/pull/176), the arms and feet ft sensors in home position measure positive Fz.

On the hip ft sensors, the Fz is negative, unfortunately, these have not been mounted yet on the real robot so we are not sure how they have to be placed.

Once we know it we eventually invert the `directionChildToParent` flag

traversaro, 2020-12-11:
Can we close this after https://github.com/robotology/icub-models-generator/pull/185 ? Or we still need to check it?

closed 2020-12-11 by Nicogene

Nicogene, 2020-12-22:
The frames of the FT in the hip seems wrong, but for now are not mounted on the real robot, and disabled from the urdf model
see https://github.com/robotology/icub-models-generator/pull/193#discussion_r547317745