robotology/icub-models #72: [iCubGazeboV3] Stange bahaviour of the left leg with the fixed model

GiulioRomualdi, 2020-11-26:
If I load the fixed model the following behavior happens

icub-models commit: https://github.com/robotology/icub-models/commit/2cfb562f04256b0f288a8d4f711579c47c59d920

[image]

Nicogene, 2020-11-26:
Hi @GiulioRomualdi !

Thanks for reporting it, the problem seems to be related to the definition of the joint, looking in the urdf we have
For the right leg
```xml
  <joint name="r_hip_roll" type="revolute">
    <origin xyz="-0.0473 -0.03310000000000001 -0.009000000000000008" rpy="0 0 0"/>
    <axis xyz="-1.0 0.0 0.0"/>
    <parent link="r_hip_1"/>
    <child link="r_hip_2"/>
    <limit effort="50000" lower="-0.17453292519943295" upper="2.007128639793479" velocity="120.0"/>
    <dynamics damping="0.06" friction="0.0"/>
  </joint>
```

And for the left leg
```xml
  <joint name="l_hip_roll" type="prismatic">
    <origin xyz="-0.04725 0.0331 -0.009000000000000008" rpy="0 0 0"/>
    <axis xyz="1.0 0.0 0.0"/>
    <parent link="l_hip_1"/>
    <child link="l_hip_2"/>
    <limit effort="50000" lower="-10.0" upper="115.0" velocity="120.0"/>
    <dynamics damping="0.06" friction="0.0"/>
  </joint>
```

Nicogene, 2020-11-26:
https://github.com/robotology/icub-models-generator/pull/183 should fix it, I checked the urdf and now they are both revolute

closed 2020-11-26 by Nicogene

GiulioRomualdi, 2020-11-26:
Thank you @Nicogene