robotology/icub-models #79: Allow possibility to not install meshes

S-Dafarra, 2020-12-11:
On systems with limited space resources, the meshes are pretty heavy and duplicated, both in the sources and in the install folder. Maybe with an option, it would be possible to avoid installing meshes, saving about 100Mb.

cc @prashanthr05 @GiulioRomualdi @traversaro

traversaro, 2020-12-11:
Just to understand, the specific case of "systems with limited space resources" you are thinking about is "pc104/icub-head", right?

S-Dafarra, 2020-12-11:
> Just to understand, the specific case of "systems with limited space resources" you are thinking about is "pc104/icub-head", right?

Yes sorry, I thought I wrote it :grin:

prashanthr05, 2020-12-11:
Related issue: https://github.com/dic-iit/component_ANA-Avatar-XPRIZE/issues/100