robotology/icub-models #87: Inserting the iCubGenova04 model into Gazebo makes it crash

S-Dafarra, 2021-03-15:
Here the video

https://user-images.githubusercontent.com/18591940/111198016-6aae7d00-85bf-11eb-84f3-4ea7a7a5cca8.mp4

I am using the latest ``devel`` version

<details>
  <summary>Here the output of Gazebo:</summary>
<p>

```
sdafarra@iiticublap104:~$ gazebo -slibgazebo_yarp_clock.so --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[DEBUG] GazeboYarpClock loaded. Clock port will be  /clock
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.84
[Msg] Loading world file [/usr/share/gazebo-11/worlds/empty.world]
[INFO] |yarp.os.Port| Port /clock active at tcp://192.168.1.84:10002/
[INFO] |yarp.os.Port| Port /clock/rpc active at tcp://192.168.1.84:10003/
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.84
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/gazebo/models/.git"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/iCub/robots/icubGazeboSim"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/BipedalLocomotionFramework"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/BlockFactory"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/HDERviz"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/HumanDynamicsEstimation"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/ICUBcontrib"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/WBToolbox"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/Wearables"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/YCM"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/bash-completion"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/cmake"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/doc"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/gazebo"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/human-gazebo"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/iCub"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/man"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/manif"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/matioCpp"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/pkgconfig"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/robotology-superbuild"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/wbc"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/robotology-superbuild/build/install/share/yarp"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/isaac/isaac-wp1/EVAmodelFiles/generatedUrdfs"
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/sdafarra/Software/isaac/isaac-wp1/EVAmodelFiles/leo_description"
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[iCub].
Warning [Model.cc:212] Non-unique names detected in XML children of model with name[iCub].
[INFO] |yarp.dev.PolyDriver| Created wrapper <multipleanalogsensorsserver>. See C++ class MultipleAnalogSensorsServer for documentation.
[INFO] |yarp.devices.ServerInertial| No ROS group found in config file ... skipping ROS initialization.
[INFO] |yarp.os.Port| Port /icubSim/inertial active at tcp://192.168.1.84:10012/
[INFO] |yarp.devices.ServerInertial| Server Inertial : no ROS initialization required
[INFO] |yarp.dev.PolyDriver| Created wrapper <inertial>. See C++ class ServerInertial for documentation.
[INFO] |yarp.dev.PolyDriver| Created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO] |yarp.os.Port| Port /icubSim/head/inertials/measures:o active at tcp://192.168.1.84:10013/
[INFO] |yarp.os.Port| Port /icubSim/head/inertials/rpc:o active at tcp://192.168.1.84:10014/
[INFO] |yarp.devices.ServerInertial| Starting server Inertial thread
[DEBUG] |yarp.devices.ServerInertial| Writing an Inertial measurement.
[INFO] |yarp.devices.AnalogWrapper| No ROS group found in config file ... skipping ROS initialization.
[... 250 lines cut ...]
[ERROR] An hardware fault occurred on joint  2  torque too big! (  4668.78  )
[ERROR] An hardware fault occurred on joint  4  torque too big! (  -9999  )
[ERROR] An hardware fault occurred on joint  5  torque too big! (  -4351.57  )
[ERROR] An hardware fault occurred on joint  4  torque too big! (  -6701.9  )
[ERROR] An hardware fault occurred on joint  0  torque too big! (  -9999  )
[ERROR] An hardware fault occurred on joint  1  torque too big! (  -5138.72  )
[ERROR] An hardware fault occurred on joint  0  torque too big! (  -8657.13  )
[ERROR] An hardware fault occurred on joint  1  torque too big! (  -9237.7  )
[ERROR] An hardware fault occurred on joint  2  torque too big! (  -9999  )
[ERROR] An hardware fault occurred on joint  0  torque too big! (  -2448.44  )
[ERROR] An hardware fault occurred on joint  2  torque too big! (  -4225.18  )
[ERROR] An hardware fault occurred on joint  3  torque too big! (  -9999  )
[ERROR] An hardware fault occurred on joint  5  torque too big! (  9999  )
[ERROR] An hardware fault occurred on joint  1  torque too big! (  -5457.92  )
[ERROR] An hardware fault occurred on joint  0  torque too big! (  2654.82  )
[ERROR] An hardware fault occurred on joint  1  torque too big! (  -6718.11  )
[ERROR] An hardware fault occurred on joint  2  torque too big! (  2622.94  )
[ERROR] An hardware fault occurred on joint  3  torque too big! (  -9481.85  )
gzclient: /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:251: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
[Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.
[INFO] |yarp.devices.ServerInertial| Closing Server Inertial...
[INFO] |yarp.devices.ServerInertial| Server Intertial thread finished
```
</p>
</details>

S-Dafarra, 2021-03-15:
This actually emerged after a mistake of mine, since I needed to use the ``iCubGazeboV2_5`` model instead. Not sure if this is an intended behavior though.

cc @Nicogene @traversaro

Nicogene, 2021-03-16:
I cannot reproduce the video, but usually, the models with the robot names are the ones without any gazebo fixes(e.g. small inertias)

traversaro, 2021-03-16:
The actual error is: 
~~~
gzclient: /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:251: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
[Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.
~~~

So I think it is some kind of problem in ogre in handling NaN that come out from numerically unstable simulation.

Nicogene, 2021-03-16:
Similar to https://github.com/robotology/icub-models/issues/53