robotology/icub-models #91: Add machinery to easily access model locations on Python

traversaro, 2021-06-05:
At the moment the models installed by this repo are supposed to be consumed by the resource location mechanism of YARP and ROS (see https://github.com/robotology/icub-models/issues/90). However, for some use cases (see https://github.com/dic-iit/ADAM/pull/1#discussion_r642348189) it is convenient to also be able to access the location of the models by a simple Python package. A similar functionality is present in https://github.com/robotology/gym-ignition-model, but having it in `icub-models` directly would permit to have directly any new model that will be designed and made available related to iCub, including iCub3 and [INAIL ergoCub](https://www.inail.it/cs/internet/comunicazione/news-ed-eventi/news/news-iit-prevenzione-robotica-video-progetto-ergocub-2021.html) robots. Borrowing the API From `gym-ignition-model` ideally it should be possible to do:
~~~python
import icub_models

print(f"Models have been installed in: {icub_models.get_models_path()}")
print(f"Available robots: {icub_models.get_robot_names()}")
print()
print("Model files:")

for robot_name in icub_models.get_robot_names():
    print(f"{robot_name}: {icub_models.get_model_file(robot_name)}")
~~~

As this repo does not depend on anything else, it would be probably a good candidate to be uploaded directly on PyPI and conda-forge. In particular adding it in conda-forge would reduce the amount of space used on the robotology channel, as the `icub-models` packages with its ~20 MB of meshes (see https://anaconda.org/robotology/icub-models/files) that in the future will increase, is one of the most heavy on the robotology channel.

traversaro, 2021-06-05:
By the way it could be convenient to have some similar machinery in form of a C++ library to find the models without depending on ROS or YARP, but to be honest I would not know at the moment how to implement easily this, as it is less obvious to do that in C++.

traversaro, 2021-06-05:
cc @Giulero @diegoferigo @GiulioRomualdi

traversaro, 2021-06-05:
Another good example of repo that makes its models available via a small Python library is https://github.com/Gepetto/example-robot-data .

traversaro, 2022-01-19:
The more the lab continue to use Python's tool, the more this issue seems important. Volunteer are more than welcome! @robotology/iit-artificial-mechanical-intelligence

traversaro, 2022-01-21:
> By the way it could be convenient to have some similar machinery in form of a C++ library to find the models without depending on ROS or YARP, but to be honest I would not know at the moment how to implement easily this, as it is less obvious to do that in C++.

https://github.com/resibots/robot_dart/pull/166 does somethings similar, but it hardcodes the installation directory in the `.hpp` (that may be indeed a simple solution).

GiulioRomualdi, 2022-02-15:
I proposed a possible implementation in #130

closed 2023-04-19 by traversaro