robotology/icub-models #92: Import bots to Unity using URDF Importer make Unity crash

Elvuis, 2021-06-09:
Hi, I'm using Unity 2020.3.1 and I tried to import robots using the URDF importer package (it is an official package from Unity Technologies and can be found here: [https://github.com/Unity-Technologies/URDF-Importer](url)), and every time, during the import process Unity crashes. I tried to import all of the robots that can be found in iCub/robots with the same results.

traversaro, 2021-06-09:
Hi @Elvuis ! 

Unfortunatly I do not know if anyone ever tried this iCub models in Unity. 

However, we used them over the years with different URDF-consuming tools without problems. As far as we know, the models are as compliant as possible to the URDF specification available at http://wiki.ros.org/urdf (beside parts that are not described in the specification). The fact that Unity is not raising any error, but it is just crashing seems to indicate that there is some bug in https://github.com/Unity-Technologies/URDF-Importer, as I would expect at least that an error is printed if a malformed URDF model is passed. Could it make sense to open an issue on https://github.com/Unity-Technologies/URDF-Importer ?

traversaro, 2021-06-09:
Just to double check, do you have an example of a URDF model that works when imported in Unity?

traversaro, 2021-06-09:
By the way, I wonder if this issue is related: https://github.com/Unity-Technologies/URDF-Importer/issues/25 . There they discuss about a maximum number of joints for the models, but I am not sure if they meant actually moving joints, are also fixed joints. If also fixed joints are considered for the limit, then we may have a problem as the iCub model has plently of fixed joints that connect fake mass-less links to represent frames (see https://discourse.ros.org/t/urdf-ng-link-and-frame-concepts/56) for a related discussion. If that is the problem, a possible strategy could be to load remove all the fake links from the iCub model, either manually or with an automatic program.

traversaro, 2021-06-25:
Hi @Elvuis, do you think it make sense to keep this issue open or the problem is Unity-related?

traversaro, 2021-06-30:
> 
> 
> Hi @Elvuis, do you think it make sense to keep this issue open or the problem is Unity-related?

Hi @Elvuis, given the lack of feedback I will close the issue, feel free to comment if you have other inputs.

closed 2021-06-30 by traversaro