robotology/icub-models #94: iCubGenova09 total mass from urdf model does not match the real robot mass

isorrentino, 2021-06-18:
The iCubGenova09 total mass from the urdf model does not match the real mass of the robot.

Here the one from the urdf model:

```
idyntree-model-info -m model.urdf --total-mass
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
``` 

and here the reality:

[image]

cc @DanielePucci @GiulioRomualdi @paolo-viceconte @S-Dafarra

DanielePucci, 2021-06-18:
CC 
@traversaro @pattacini @maggia80

pattacini, 2021-06-18:
@salvi-mattia we should run a second check on all these weights.

DanielePucci, 2021-06-18:
CC @S-Dafarra

traversaro, 2021-06-18:
If you want to check which commit introduced the problem, you can run `idyntree-model-info --total-mass -m model.urdf` for each commit in the history if `icub-models` to understand at which point in the history  the problem was introduced.

For a given icub-models commit, you then can check the corresponding icub-model-generator commit in the commit message, see https://github.com/robotology/icub-models/commit/6d760251473396133daaf75ae7a3b27137da607f . From that, we can understand what caused the regression.

pattacini, 2021-06-18:
Very nice feature, indeed!

GiulioRomualdi, 2021-06-19:
I wrote this script
```bash
#!/bin/bash

RED="\e[31m"
ENDCOLOR="\e[0m"

commits=(`git rev-list --ancestry-path cd4b2a6608cea70d0c455b68e18f414cb0820c0c..6d760251473396133daaf75ae7a3b27137da607f`)
for commit in ${commits[@]}
do
    echo -e "${RED} ${commit} ${ENDCOLOR}"

    git checkout ${commit}
    idyntree-model-info -m model.urdf --total-mass #
done
```

you should run it inside `icub-models/iCub/robots/iCubGenova09`. I took the first and last commits from the history of the file https://github.com/robotology/icub-models/commits/master/iCub/robots/iCubGenova09/model.urdf

Here the output  
```
 6d760251473396133daaf75ae7a3b27137da607f 
Previous HEAD position was 023acfc Automatic build. Travis build:390
HEAD is now at 6d76025 Automatic build. GitHub Actions build: 882676287 icub-model-generator commit: robotology/icub-models-generator@1472a40ae34b0601b82ed0d60406ab3300347e2d urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@1eb31b4e2d86e091c59cbe3cfc62d1e5265d7d61
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
 d49a788e03c00cba44512bc1007edd90c924a7ca 
Previous HEAD position was 6d76025 Automatic build. GitHub Actions build: 882676287 icub-model-generator commit: robotology/icub-models-generator@1472a40ae34b0601b82ed0d60406ab3300347e2d urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@1eb31b4e2d86e091c59cbe3cfc62d1e5265d7d61
HEAD is now at d49a788 Automatic build. GitHub Actions build: 773585480 icub-model-generator commit: robotology/icub-models-generator@8e36592b8124e49c92a07a553d11d466be2d3611 urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
 fcc20f76dfe526ca0d4302e6a5bed1240ba77671 
Previous HEAD position was d49a788 Automatic build. GitHub Actions build: 773585480 icub-model-generator commit: robotology/icub-models-generator@8e36592b8124e49c92a07a553d11d466be2d3611 urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
HEAD is now at fcc20f7 Automatic build. GitHub Actions build: 689389661 icub-model-generator commit: robotology/icub-models-generator@921c4cbd5eefd94e95437ad3bb01793aff3ace68 urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
 192c4a1fc0eeaa54c7cc222e99a1c10e4a0704a7 
Previous HEAD position was fcc20f7 Automatic build. GitHub Actions build: 689389661 icub-model-generator commit: robotology/icub-models-generator@921c4cbd5eefd94e95437ad3bb01793aff3ace68 urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
HEAD is now at 192c4a1 Automatic build. GitHub Actions build: 589039128 icub-model-generator commit: robotology/icub-models-generator@20a9bbcfa6e962c44be22a30c531724a0227fb7e urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
 e8eb113f81fa8d4d76ae899c4eb5b2c0baafa62c 
Previous HEAD position was 192c4a1 Automatic build. GitHub Actions build: 589039128 icub-model-generator commit: robotology/icub-models-generator@20a9bbcfa6e962c44be22a30c531724a0227fb7e urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
HEAD is now at e8eb113 Automatic build. GitHub Actions build: 518010635 icub-model-generator commit: robotology/icub-models-generator@5090736272aec2e339a9eb71b21d7f00744f991e urdf_parser_py commit: ros/urdf_parser_py@31474b9baaf7c3845b40e5a9aa87d5900a2282c3 simmechanics-to-urdf commit: robotology/simmechanics-to-urdf@f6055ab8d627c45138ddde46532ea42a74b48caa
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type depth
[WARNING] SensorElement :: setAttributes : iDynTree does not support sensor of type camera
The total mass of model is 58.458 Kg.
 b79c089d668ce76474580d0cd3b6b3e984b3cead 
[... cut ...]
HEAD is now at 18b438e Fixed multiple generation of fixed models and add uninstall target.
The total mass of model is 19.9045 Kg.
 29f5fddad7f9da4dbebc67e2fac13f9dcb2aaceb 
Previous HEAD position was 18b438e Fixed multiple generation of fixed models and add uninstall target.
HEAD is now at 29f5fdd Automatic build. Travis build:395
The total mass of model is 19.9045 Kg.
 023acfcbe40bf7b6887da1ca9fba30db4b5dd9e6 
Previous HEAD position was 29f5fdd Automatic build. Travis build:395
HEAD is now at 023acfc Automatic build. Travis build:390
The total mass of model is 19.9045 Kg.
```

pattacini, 2021-06-19:
Thanks @GiulioRomualdi for checking.
Thus, we haven't had a robot model weighting 51 kg.
@salvi-mattia is on it.

S-Dafarra, 2021-06-29:
Related issues/discussions:
- https://github.com/robotology/icub-tech-support/issues/98
- https://github.com/robotology/community/discussions/61
- https://github.com/robotology/community/discussions/56

salvi-mattia, 2021-07-23:
At this point we have a 3kilos delta between the CAD and RL robot.
less than 6%

As me and @DanielePucci said last time
we would like to proceed creating the new simulation model in cad and export a new version of URDF

at this point we would like to remove this three kilos equally from the 7 big assemblies of the robot [chest, upper legs, lower legs, upper arms, forearms]

cc @S-Dafarra @fiorisi 

[image]
[image]

DanielePucci, 2021-07-23:
Great, it makes sense to distribute these three kilos around! In fact, we checked more or less the CoM with @GiulioRomualdi and @isorrentino and more or less the CoM was where the model was suggesting. This CoM check, however, could be a further point to be investigated later on, having now the objective of having URDF weght = scale weight 🚀

pattacini, 2021-08-06:
A PR alleviating this weight mismatch is expected to land soon.
See https://github.com/robotology/icub-models-generator/pull/207.

Thanks @salvi-mattia @mfussi66 @ale-git!

closed 2021-08-31 by pattacini

DanielePucci, 2021-12-07:
CC @CarlottaSartore @lrapetti @S-Dafarra