robotology/icub-models #96: Unstable behavior of left_hand_mk3

gabrielecaddeo, 2021-07-16:
Hi everyone,

I am trying to use the new hand model (i.e. `left_hand_mk3`) from the devel branch. However, as soon as I insert it in Gazebo it starts "jumping" and, after some iterations, it collapse on its own.

I am attaching a video which shows the aforementioned behavior.

https://user-images.githubusercontent.com/49904924/125916196-6b160d20-058c-4d29-9d8a-9720fce248f2.mp4

traversaro, 2021-07-16:
Hi @gabrielecaddeo ! First of all it would be great if you could provide the exact version of the software you are using beside icub-models, including Gazebo, YARP and gazebo-yarp-plugins and the physics options that you used (the default ones?). After that, probably @ale-git is the right person to ask about this, as I guess he did some tests on that model.
 
As a general comment, I can say that the default physics settings of Gazebo with the default physics engine (`ode`) make it quite challenging to simulate/integrate stiff mechanical systems ("stiff" in this context is used with the meaning used in integration of differential equations, see https://en.wikipedia.org/wiki/Stiff_equation ). 

The possible way to proceed in this cases are:
* Change the physics engine (did you already tried with `dart`)?
* Change the physics engine parameters to reduce the numerical error (in the case of `ode`, you can read about the parameters there: https://gazebosim.org/tutorials?tut=physics_params&cat=physics ). In general, the easist option is to first decrease the `max_step_size`, as all physics engine used by Gazebo used fixed-step integrator without any form of error control, so decreasing 
* Change the control parameters: the gain of the PID controllers also contribute to the numerical stiffness of the system, so decreasing them can help.
* Increase the joint damping. I am not sure which joint damping has been used in this model, probably @ale-git can provide some info, but tipically increasing the damping decreases the stiffness of the system.
* Increase the link inertia. This is indeed the strategy we are still currently using on iCub models to ensure that they run fine on Gazebo with the default settings. We do that by providing modified models, but this proved out to be difficult to mantain (see https://github.com/robotology/icub-models/issues/33). Ideally for future models we would like to avoid that, eventually increasing inertia at runtime via appropriate plugins, but without shipping modified models.

gabrielecaddeo, 2021-07-19:
Sorry, on friday the network was down and then I forgot to answer. It works with DART, thanks. I was wondering if you have in mind how to make it work with ODE. I know the advantages of DART with respect to ODE, but I want to use the hand for contact detection and force reading and DART does not behave great (e. g. https://github.com/dartsim/dart/issues/1425). I tried your suggestion with ODE (like changing `max_step_size` or increasing link inertia and joint damping) but it still does not work.

traversaro, 2021-07-19:
@ale-git did you do any experiment with this model? Which settings did you used?

traversaro, 2021-07-19:
@gabrielecaddeo did you tried to fix the hand to some fixed base, so that it does not fall on the ground?

gabrielecaddeo, 2021-07-19:
I just tried, it collapses as before but without jumping 

https://user-images.githubusercontent.com/49904924/126194282-a5149ea8-48ed-49f1-8760-d2b56c321791.mp4

traversaro, 2021-07-19:
Can you try to elevate it a bitw.r.t. to the ground? As it is now, the hand is spawned completly in collision with the ground plane, and this is probably creating problems.

gabrielecaddeo, 2021-07-20:
I tried yesterday night, seems no improvement. By the way, thanks for the help you are giving me

traversaro, 2021-07-20:
> I tried yesterday night, seems no improvement. I was thinking about creating another dummy link and then fixed it to the ground. 

No additional dummy links should be necessary, and actually it can create even more numerical problems. How are you adding the fixed joint? That could be the problem. This is what we do historically for iCub models: https://github.com/robotology/icub-models/blob/master/fixed_model.sdf.in .

> By the way, thanks for the help you are giving me

You are welcome!

gabrielecaddeo, 2021-07-20:
Okay now it works, I fixed directly the root link  `l_hand_palm_link` in the urdf file of the model. However, I'll try to use DART as physics engine for my project, but knowing that ODE works is good. Are there any possibilities to make it work even without fixing it to the world? If something else come to your mind tell me and I will try

traversaro, 2021-07-20:
> Are there any possibilities to make it work even without fixing it to the world?

I think probably in that case we may need to tune the contact parameters or use simpler collision meshes, as they may be the one that are creating the contact force that make the model simulation explode.