robotology/icub-models #104: Design of the urdf model of the left_hand_mk3 seems different from the real design

gabrielecaddeo, 2021-09-10:
Hi,
I am working with the urdf model of the `left_hand_mk3`.  While working with it I noticed that there is something strange going on with the thumb opposition joint.

As you can see in the following

[image]

in the real hand it is possible to reach a configuration in which the thumb and the index superimpose one each other.

On the other hand, in the simulation

[image]

that behavior is not possible (please notice that in the above picture the thumb opposition is already at its maximum of approximately ~62 degrees. The same applies for the index adduction/abuction that is at 10 degrees.)

traversaro, 2021-09-10:
fyi @pattacini @fiorisi

pattacini, 2021-09-11:
Thanks @gabrielecaddeo for reporting 👍🏻 

@ale-git is busy these days with the development of a FW release to address an important upcoming demo.
We might have some time to look into it starting with our next sprint (Mon 20).

traversaro, 2021-09-13:
In the meanwhile, some tips for @gabrielecaddeo so that you can continue to debug on your own and prepare useful information for domain experts.

First of all, which version (tagged version or git commit checkout if you are not in a released version) of `icub-models` are you using? Which version of YARP? Which version of Gazebo? Which version of gazebo-yarp-plugins?

> it is not possible to reach a configuration in which the thumb and the index superimpose one each other (please notice that in the above picture the thumb opposition is already at its maximum of approximately ~62 degrees. The same applies for the index adduction/abuction that is at 10 degrees).

Can you report this information using the precise identifier of joint names, i.e. the one that you find in the URDF? Using ambigous names may lead to confusion that make debugging harder.

> On the other hand, in the real hand it is possible to reach this configuration, as I tried to do it today.

If you are able to reproduce this configuration with the real hand, can you report in the issue the exact configuration of the hand (i.e. in a table with joint/axis names and their values?). After that, you can try to reproduce the same configuration in simulation, for example using `yarpmotorgui`. If you try that, I guess that there are two possible outcomes:
* You are able to have in the simulated robot the same configuration, but in the real hand the fingertip do not touch. This may indicate that there is something wrong (i.e. not matching the real hand) **in the simulated URDF kinematic model** or in the **coupling laws** added in https://github.com/robotology/gazebo-yarp-plugins/pull/469 .
* You are **not** able to reproduce the same configuration in simulation. This may indicate that something is wrong in the joint/axis limits.

gabrielecaddeo, 2021-09-13:
> First of all, which version (tagged version or git commit checkout if you are not in a released version) of `icub-models` are you using? Which version of YARP? Which version of Gazebo? Which version of gazebo-yarp-plugins?

I am using `YARP` 3.4.5+6-20210610.3+gitfdc109fa6, `Gazebo 11.8.1`, the last commit of `icub-models` (devel branch, b9d4d49fe4571d5823cb1b59766bcdef0326945c), `gazebo-yarp-plugins` 0d18d2ed4e20c6cfb751e976171781df0deb9acd v3.6.1.

> Can you report this information using the precise identifier of joint names, i.e. the one that you find in the URDF? Using ambigous names may lead to confusion that make debugging harder.

In my opinion, the problem is with the `l_thumb_oppose joint` (JOINT 0) that reaches just 62 degrees. The other joint I was talking about is the `l_thumb` (JOINT 2)

> If you are able to reproduce this configuration with the real hand, can you report in the issue the exact configuration of the hand (i.e. in a table with joint/axis names and their values?)

I am not sure I can control the real hand at the moment, because it is under test on the first floor. I went there and I asked if I could move the fingers with my hand to see the limits of the real joints. I can try to go there and  ask if i can move it with the software.

traversaro, 2021-09-13:
Thanks @gabrielecaddeo !

> I am not sure I can control the real hand at the moment, because it is under test on the first floor. I went there and I asked if I could move the fingers with my hand to see the limits of the real [...]

Sorry, I misunderstood then. I think at some point there was a working prototype, probably you can ask, otherwise we can wait for @ale-git input.

traversaro, 2021-09-13:
To do some tests, you can try to increase the software and hardware limits in:
Software Limits: https://github.com/robotology/icub-models/blob/v1.21.1/iCub/conf_icub3/gazebo_icub_left_hand_thumb.ini#L68

The hardware limits should be defined in https://github.com/robotology/icub-models/blob/v1.21.1/iCub/robots/left_hand_mk3/model.urdf#L261, but I am bit confused as there I can only find the `l_hand_thumb_proximal_joint`, `l_hand_thumb_distal_joint` and `l_hand_thumb_rotation_joint` joints, no reference to `l_thumb` and `l_thumb_oppose` joints.

traversaro, 2021-09-13:
I think I got it. The actual software limits (for the coupled controlled axis, that have indeed the correct `l_thumb`, `l_thumb_oppose` names) are defined in https://github.com/robotology/icub-models/blob/c73aa1d7e42e571c0b5b9f12043a07f9a1377e3f/iCub/conf_left_hand_mk3/left_hand_phys.ini#L24 , so feel free to try to change them or the hardware limits.

ale-git, 2021-09-13:
Hi @gabrielecaddeo,
the Mk3 hand is designed with an additional DOF

[image]

so that the pinch grasp is achievable. In your screenshot the thumb (second joint in yarpmotorgui) is not rotated, i.e. it seems in 0 degrees position, while it can reach -70 degrees.

gabrielecaddeo, 2021-09-13:
> so that the pinch grasp is achievable. In your screenshot the thumb (second joint in yarpmotorgui) is not rotated, i.e. it seems in 0 degrees position, while it can reach -70 degrees.  
Hi @ale-git .
Since it is a negative rotation, it is even worst for the grasp, because the thumb would be orthogonal to the index.
I attach a pic here
[image]

ale-git, 2021-09-13:
Well, also in human hand thumb and index aren't perfectly opposed but kinda lateral in pinch grasp. Please consider that you don't have to go to -70, but you have to rotate just the amount you need. Btw, can you post a picture of the real hand performing the grasp you desire? Maybe there is an offset in the rotation limits between the real hand and the model.

gabrielecaddeo, 2021-09-13:
I did not post the picture of the real hand since this is public, but I will now, sorry.
[image]

Clearly it is not perfectly "parallel" but it is very close to the desidered configuration.

ale-git, 2021-09-13:
Maybe the thumb opposition limit is changed @marcellosavoldi ?

marcellosavoldi, 2021-09-20:
Hi guys... sorry for late.
The limit range is not changed .
I confirm that the opposition of the index and thumb is not parallel, but the fingertip of the thumb touches the index when it is slightly inclined.
[image]
[image]

metacarpus 0-80°
thumb rotation 0-70° ( to be honest , I dont remember the zero position... could be +35,-35° ... or 0,-70? )

ale-git, 2021-09-20:
The thumb metacarpus range in the configuration file is smaller:

[image]

We must increase it to 80 degrees, and the difference between the URDF and physical model should vanish.

traversaro, 2021-09-20:
I guess this value should be updated in two places:
* https://github.com/robotology/icub-models-generator/blob/v1.21.1/simmechanics/data/components/left_hand_mk3/simmechanics2urdf_joints_configfile.csv#L2
* https://github.com/robotology/icub-models-generator/blob/v1.21.1/simmechanics/data/components/left_hand_mk3/conf/left_hand_phys.ini#L25

Anyone interested is welcome to submit a PR, thanks!

pattacini, 2021-09-29:
Fixed in https://github.com/robotology/icub-models-generator/pull/211.

closed 2021-09-29 by pattacini

traversaro, 2021-09-29:
Thanks @mfussi66 !