robotology/icub-models #106: `velocityControlImplementationType` missing parameter for iCubGazeboV3

GiulioRomualdi, 2021-09-23:
This parameter is required after the gazebo yarp plugin v4.0.0

mebbaid, 2021-11-11:
any updates on this issue ?

GiulioRomualdi, 2021-11-11:
This should fix the issue
 https://github.com/robotology/icub-models-generator/pull/212

Could you try with the devel branch?

mebbaid, 2021-11-11:
@traversaro this maybe the issue that the Jonathan from Australia was having  (he mentioned this during the ergocub update meeting). Couldn't tag him because i didn't find him on this site.

GiulioRomualdi, 2021-11-11:
Given

closed 2021-11-11 by GiulioRomualdi

lrapetti, 2021-11-12:
> @traversaro this maybe the issue that the Jonathan from Australia was having (he mentioned this during the ergocub update meeting). Couldn't tag him because i didn't find him on this site.

@Woolfrey

traversaro, 2021-11-12:
Thanks a lot @mebbaid, I think indeed that it is! https://github.com/robotology/icub-models/commit/6239efecaf2d19368d14ad5a322ec0afc767c8a8 (and the related PR https://github.com/robotology/icub-models-generator/pull/212) fixed it in devel, but probably @Woolfrey downloaded either master or released version of icub-models. Let's release a new version of icub-models so users can get a working iCub3 models without a lot of hassle.

traversaro, 2021-11-12:
Hopefully we can do a 1.22 release (see https://github.com/robotology/icub-models/pull/109) to avoid confusion in all users.

traversaro, 2021-11-12:
> 
> 
> Hopefully we can do a 1.22 release (see #109) to avoid confusion in all users.

Done: https://github.com/robotology/icub-models/releases/tag/v1.22.0 .

traversaro, 2021-11-12:
> 
> 
> > Hopefully we can do a 1.22 release (see #109) to avoid confusion in all users.
> 
> Done: https://github.com/robotology/icub-models/releases/tag/v1.22.0 .

I also updated the `LatestReleases` file in the robotology-superbuild: https://github.com/robotology/robotology-superbuild/pull/920 . So the next conda package generation will get the updated icub-models, let me know if there is anything else that we could do to make sure that this issue is fixed without the need for the user to manually switch branches or do stuff like that. @mebbaid @Woolfrey

mebbaid, 2021-11-12:
@traversaro it seems that this fixes the initial issue and the `iCubGazeboV3` starts correctly.

Woolfrey, 2021-11-18:
I saw this; thanks!

Github notifications are being sent to my gmail instead of my iit email.