robotology/icub-models #108: Upper-body model with new wrist and hand

MatteoPerotto, 2021-11-03:
Hi all, 

me and @xEnVrE are going to test the kinematic of the new iCub hand.
We need to run simulations of a complete and up-to-date upper-body model (torso, head, arm). This has to be representative of the actual hardware configuration, with the new wrist and hand attached to the upper body.
 
Is there any plan to release an automatically generated assembly of the aforementioned parts? 

Thank you

marcellosavoldi, 2021-11-03:
currently there is  no CAD model of the Upperbody + Hands assembly (not yet planned).
Honestly, there is no right hand model either (only the left hand was designed and tested for mechanical testing).
Starting from the CAD to generate a simulable model, I believe that for now it is not possible .... we have to work on it!

PS: The hand we are physically testing is the 4-finger version, but the new 5-finger hand is work in progress.
The new 5-finger hand moves the ring finger and the pinky at the same time (powered by a single motor), so I don't think there are any problems testing the 4-finger version for now.
what do you think @fiorisi ?

[image]

[image]

xEnVrE, 2021-11-03:
Thank you for the answer @marcellosavoldi 

> so I don't think there are any problems testing the 4-finger version for now.

Indeed, our plan was to work with the 4-fingers version for now.

> Starting from the CAD to generate a simulable model, I believe that for now it is not possible

Thank you for this information. Given that the URDF of both the 4-fingers hand and of the wrist are both available [here](https://github.com/robotology/icub-models/tree/master/iCub/robots/left_hand_mk3) and [here](https://github.com/robotology/icub-models/tree/master/iCub/robots/left_wrist_mk2) respectively, do you think we can try to attach them manually in a newer URDF (just for our purposes)?

I also have a question regarding the upper body the 4-fingers hand + wrist is attached to. Is it (up to the forearm) the same of a standard iCub 2.5?

fiorisi, 2021-11-04:
> Thank you for this information. Given that the URDF of both the 4-fingers hand and of the wrist are both available here and here respectively, do you think we can try to attach them manually in a newer URDF (just for our purposes)?

Yes, currently we released only the URDF models of the left hand mk3 and left forearm mk2.
Probably we will update the models (and mirror the assembly) once the 5 fingers hand has been finalized.

> I also have a question regarding the upper body the 4-fingers hand + wrist is attached to. Is it (up to the forearm) the same of a standard iCub 2.5?

If you mean the mechanical interface, then yes is the same of the previous forearm.
[image]

xEnVrE, 2021-11-04:
Just to clarify, our request stems from the fact that we would like to reproduce the actual physical setup available in the laboratory composed of a robot upper body, a sample of the new wrist and a sample of the new hand (the 4-fingers variant).

> If you mean the mechanical interface, then yes is the same of the previous forearm.

Ok. We will try to manually compose the three parts then and report back in case we have doubts about the links and their conventions.

MatteoPerotto, 2021-11-08:
I'm merging the wrist and hand urdf files. The root element of the hand file is the `hand_palm_link`. In order to have a valid tree, it is necessary to create a joint that connects the `l_platform` link (the wrist mechanism) to the palm link. 

Can you please give us the values of distances and angles of the rigid transformation between the palm reference frame and the one located where wrist is attached to the hand?

Hope I've been clear enough, I put also two pictures. 

Thanks

[image]
[image]

marcellosavoldi, 2021-11-09:
@ale-git , @fiorisi  
do we have any datas about the wrist in any issue?

fiorisi, 2021-11-10:
> @ale-git , @fiorisi
do we have any datas about the wrist in any issue?

No, @marcellosavoldi can you just measure the transformation between the two frames highlighted in [this comment](https://github.com/robotology/icub-models/issues/108#issuecomment-963165962)?

MatteoPerotto, 2021-11-10:
Dear @marcellosavoldi, in order for you to save time, I'm available for a brief face-to-face meeting where you measure the transformation and I insert it on the fly in my urdf file. This avoids me to ask you further information in case the merging operation results to be, for whatever reason, incorrect.