robotology/icub-models #114: iCub Model Collapses in Compliant Interaction Mode

Woolfrey, 2021-12-29:
Hello!

I've been running the iCub v3 model in Gazebo with the joints in velocity mode:
```
this->mode->setControlMode(i, VOCAB_CM_VELOCITY);
```

When I set the interaction mode to compliant, the model collapses:
```
this->interaction->setInteractionMode(i, yarp::dev::VOCAB_IM_COMPLIANT);
```
[image]

When I set it to _VOCAB_IM_STIFF_ it works fine. I'm not sure if the problem is my end, with the Gazebo simulation, or something else.

The problem is not urgent. I want to develop a Cartesian impedance controller, so I will likely switch to torque mode soon.

- Jon

traversaro, 2021-12-29:
Thanks @Woolfrey ! I guess there are some not properly set gains, I can check when I am back close to a keyboard.

Woolfrey, 2021-12-29:
Actually, now that I think about it, stiffness relates to the position feedback gain. I'm running in velocity mode, so there is no position feedback!
[image]
That might explain it. It doesn't really make sense to have compliance with a velocity controller.

I should probably switch over to torque control anyway.

traversaro, 2022-01-13:
Hi @Woolfrey, any news on this? Did you investigate the problem?

Woolfrey, 2022-01-13:
Hi @traversaro, I haven't had time to look in to it further. I am working on re-doing my code using the iDynTree libraries so I can implement a proper force/torque controller.

Do you have any insight on the problem? If it is an easy fix I can test it out. But my long term goal is a proper torque controller, so all this work in velocity mode will be obsolete.

If it's too much effort, feel free to close this issue!

traversaro, 2022-03-18:
> Do you have any insight on the problem? If it is an easy fix I can test it out. But my long term goal is a proper torque controller, so all this work in velocity mode will be obsolete.

Sorry @Woolfrey, I had missed your question. Probably what is happening is that internal impedance loop has some wrong gains

Woolfrey, 2022-03-19:
Hi @traversaro, quite alright! I'm running in torque mode now using iDynTree so it's technically no longer an issue for me.

You can close this issue if you like, but it may also be a problem for someone else in the future.