robotology/icub-models #115: Changing Joint Limits in Simulation?

Woolfrey, 2022-01-05:
I want to perform a certain action with the iCub in Gazebo, but the angle at the elbow is too large due to joint limits:

[image]

I have tried altering it with the `yarp::dev::iControlLimits` class:
```
this->limits->setLimits(i,lower,upper);
```
But if I force the joint beyond its default limits it will not move, and I get the following error message:
```
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  3  torque too big! (  -9999  )
```
Is it possible to alter this constraint, just for simulation purposes?

Thanks in advance,

Jon

traversaro, 2022-01-10:
The `setLimits` interface can only change the so-called **software** limits, i.e. the limits imposed by the joint-level controller, both on the simulator and on the real robot. What can't be changed by the setLimits (both on simulation and on the real robot) are the **hardware** limits, i.e. the configuration where there it is not possible to move the joints due to some kind of hardware hardstop. Indeed, I am not sure if HW fault is the right behavior when setLimits outside the hardware limits are asked, and for that I opened https://github.com/robotology/icub-models/issues/116 .

On the Gazebo-simulated robot, to change the hardware limits you need to modify the URDF model of the robot, i.e. the `lower` and `upper` limits of the `limit` tag, see https://github.com/robotology/icub-models/blob/c73aa1d7e42e571c0b5b9f12043a07f9a1377e3f/iCub/robots/iCubGazeboV3/model.urdf#L636 for `r_elbow`. If you modify the hardware limit, then you need to remember also to modify the corresponding default software limit, i.e. the value in https://github.com/robotology/icub-models/blob/c73aa1d7e42e571c0b5b9f12043a07f9a1377e3f/iCub/conf_icub3/gazebo_icub_right_leg.ini#L69 .

In the case you are generating the models from icub-model-generator, the file to modify is https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/ICUB_3_joint_all_parameters.csv#L30 .

Woolfrey, 2022-01-10:
Thanks for the information, @traversaro.

I wasn't sure if the `setLimits` function would also affect the Gazebo simulation, or if the constraints were set somewhere else. But modifying the URDF makes sense. I will give it a try!

Jon.

traversaro, 2022-01-13:
Hi @Woolfrey, I will close this issue as the information has been provided. Feel free to re-open or open a new one if you still have problems.

closed 2022-01-13 by traversaro