robotology/icub-models #119: Clarifications on joint limits of eyes and finger adduction/abduction

xEnVrE, 2022-01-13:
These are the limits for the right hand finger adduction/abduction

https://github.com/robotology/icub-gazebo/blob/14a078e49968c2130d64b8d5635d6f302035d67b/icub/conf/gazebo_icub_right_hand_finger.ini#L68-L71

On the real robot, these correspond to a single DoF spanning approximately 60.0 degrees. Please also consider that the joint associated to the middle finger is not actuated.

Hence I am expecting something like

```
[LIMITS] 
 jntPosMax 20.0 0.0 20.0 20.0 
 jntPosMin 0.0 0.0 0.0 0.0 
 jntVelMax 100.0 100.0 100.0 100.0 
```
 
because, when the hand is totally open, the best that we can do is to assign one third of the range to each joint. The link associated to the middle finger, instead, does not move.

As far as I understand, the `controlboard` plugin within `gazebo-yarp-plugins` uses the limits to decide the limits of the trajectory generator. The trajectory is associated to the controlled DoFs (i.e. the ones that we can control via, e.g., the `yarpmotorgui`) and is remapped to the physical joints simulated within `Gazebo` using coupling handlers (see [here](https://github.com/robotology/gazebo-yarp-plugins/blob/3fa2c45a4b8a63ef15c524462124b505f29b3ed9/plugins/controlboard/include/yarp/dev/ControlBoardDriverCoupling.h)). 

Right now, the effect of the coupling on the limits is not handled and [this](https://github.com/robotology/gazebo-yarp-plugins/pull/472) PR is trying to address this issue. Without this PR, the limits indicated above will result in a minimum possible value for the joint `r_hand_finger` of `0.0` and a maximum possible value of `20.0`. Of course, this is wrong.

The same apply for the left hand.

I also have doubts for the `eyes`. These are the joint names

https://github.com/robotology/icub-gazebo/blob/14a078e49968c2130d64b8d5635d6f302035d67b/icub/conf/gazebo_icub_head.ini#L27

with associated limits

https://github.com/robotology/icub-gazebo/blob/14a078e49968c2130d64b8d5635d6f302035d67b/icub/conf/gazebo_icub_head.ini#L72-L75

The physical joints (excluding the neck) are `eyes-tilt`, `left-eye` and `right-eye` while the commanded DoF are the `eyes-tilt`, `eyes-version` and `eyes-vergence`. Hence, I am expecting that the limits associated to the physical joints are reported while `(-30.0 30.0)` is indeed the typical range for the `eyes-version` and `(0.0 50.0)` is the typical range for the `eyes-vergence`.

Is any specific reason for this choice of the limits?

xEnVrE, 2022-01-19:
@traversaro since the issue has been moved here, I think it can also be closed as the only model with eyes and fingers, if I am not wrong, is [iCubGazeboV2_5_visuomanip](https://github.com/robotology/icub-models/tree/master/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip) for which proper limits have been set.

traversaro, 2022-01-19:
Great!

closed 2022-01-19 by traversaro