robotology/icub-models #120: Make sure ARE is working in Gazebo simulation

traversaro, 2022-01-13:
Several users are experiencing issues that (I think) are related to not properly tuned joint-level PID position loops of the iCub model in Gazebo, see:
* https://github.com/robotology/QA/issues/155
* https://github.com/robotology/icub-gazebo/issues/37
* https://github.com/robotology/icub-gazebo/issues/35
* https://github.com/robotology/icub-gazebo/issues/31

We should tune the position gains with respect to some clearly defined scenario, such as the ARE one described by @frank-foerster in https://github.com/robotology/icub-gazebo/issues/37 . For reference we should document on which scenario we tuned the gains, such that in the future we can simplify the process of re-tuning the gains for different robots (such as R1).

pattacini, 2022-01-13:
Most of the basic components of ARE (Cartesian Controller, Gaze Controller, hand controller) are working nicely now as demonstrated in https://github.com/robotology/icub-gazebo-grasping-sandbox.

Never tested ARE per se though.

pattacini, 2022-01-13:
In particular, I've never checked whether WBD can run just fine.