robotology/icub-models #124: Fix iCubGazeboV2_5_visuomanip colors

traversaro, 2022-01-28:
The model `iCubGazeboV2_5_visuomanip` added in https://github.com/robotology/icub-models/pull/42 has strange "random" colors. For automatically generated models, the issue will be fixed by https://github.com/robotology/icub-models-generator/pull/217, but for `iCubGazeboV2_5_visuomanip` if we want to fix the issue we need to it specifically for this model.

fyi @xEnVrE @pattacini

pattacini, 2022-01-28:
"Random" in which sense? Isn't the "standard" anonymous whitish color?

>if we want to fix the issue we need to it specifically for this model.

Fine with me!
Since @xEnVrE authored it, I'll leave to him the final choice.

xEnVrE, 2022-01-28:
Hi @traversaro @pattacini,

fine with me. I imagine that I could check how the automatic pipeline changes the materials in the automatically generated models and apply the changes in the manually generated model.

@traversaro do you think this solution is doable? (I can take care of it.)

traversaro, 2022-01-28:
> "Random" in which sense? Isn't the "standard" anonymous whitish color?

Actually, not. The whitish color we were mostly used to is due to a bug in URDF parsing that was both in iDynTree and in Classic Gazebo, but if you displayed the model in a ROS tool like RViz you were always getting it as an "Arlecchino". @GiulioRomualdi is fixing the issue in iDynTree (see https://github.com/robotology/idyntree/pull/961), so we were going to see an arlecchino robot in iDynTree, but in Classic Gazebo we will continue to see a whitish color.

traversaro, 2022-01-28:
> Hi @traversaro @pattacini,
> 
> fine with me. I imagine that I could check how the automatic pipeline changes the materials in the automatically generated models and apply the changes in the manually generated model.
> 
> @traversaro do you think this solution is doable? (I can take care of it.)

Ok!

xEnVrE, 2022-03-11:
I made the changes.

Some notes:
- I took the colors from the latest `icub-models` (not sure why the red color was picked up for the entire robot, however I will leave it in order to have coherent colors w.r.t. the automatically generated colors)
- I had to change the color of the part indicated with the green arrow. It is a metallic part, hence it should not be red, I guess.
- The top and palm cover of the hand are black as in real robots
- The fingertip should be black, however the mesh is shared with the last part of the finger, hence I could not make it better than that.
- The following were generated using the meshcat-based visualizer provided [here](https://github.com/robotology/idyntree/blob/master/examples/python/MeshcatVisualizerExample.ipynb)

[image]

[image]

### Eyes bulbs

I will use the `idyntree-model-viewer` to show you the eyes as the meshcat-based one does not support spheres and cylinders, that are used for the visual description of the eyes.

[image]

If this configuration looks good I can open a PR. @traversaro @pattacini 

Btw, I noticed that using the idyntree visualizer the colors of hand and fingers are not taken into account. Is this because they are `dae`s instead of `stl`?

pattacini, 2022-03-11:
Fine with me 👍🏻

traversaro, 2022-03-12:
> (not sure why the red color was picked up for the entire robot, however I will leave it in order to have coherent colors w.r.t. the automatically generated colors)

Basically the one that do the modification get to choose the color, see https://github.com/robotology/icub-models-generator/pull/217#issuecomment-1024312288 . :D 

> Btw, I noticed that using the idyntree visualizer the colors of hand and fingers are not taken into account. Is this because they are daes instead of stl?

Yes, in the iDynTree visualizer `.dae`  files can specify their own color inside the file itself, so the URDF color file is not taken into account. This is also the behaviour of Gazebo if I recall correctly.

PR #139 (2022-03-14): Fix material colors of iCubGazeboV2_5_visuomanip 
As per the title, this PR fixes the colors of the mode `iCubGazeboV2_5_visuomanip`

Fix #124
closed 2022-03-14 by traversaro