robotology/icub-models #127: iCubGenova09 hip pitch limits differ from robot configuration

lrapetti, 2022-02-15:
Currently the `hip_pitch` upper joint limits in the iCubGenova09 model are set to [115 degrees](https://github.com/robotology/icub-models/blob/ab11b8cb8799b85a8805be95e3d66e7c6c7585ff/iCub/robots/iCubGenova09/model.urdf#L51) (2.007128639793479 rad).

This value looks to be much different from the actual limits in [robots configuration](https://github.com/robotology/robots-configuration). In particular, that value is matching previous values of joint limits, but they have been changed in https://github.com/robotology/robots-configuration/commit/1a88059334cac87d83a2816864cbc94c2e3ae699, and are now set to be:
- 94 degrees in the `mechanical`
- 90 degrees in the `motorControl`

S-Dafarra, 2022-02-15:
As you can see in https://github.com/robotology/robots-configuration/blob/8a37d27138f722b0fc35084840c70a8637355ecd/iCubGenova09/calibrators/left_leg-calib.xml#L25, actually 93deg is the value of the hard stop. In this way, when ar 0deg, the leg is straight.

lrapetti, 2022-02-15:
We should then change the limit in the urdf as well (via [`icub-models-generator`](https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/conf/gazebo_icub_right_leg.ini#L71)).  Not sure thou what value we should use (94, 93, 90?)

traversaro, 2022-03-02:
@pattacini do you have any idea on the "right" limits?

ale-git, 2022-03-02:
About the calibration value we set it to 93 because this was the value shown by the yarpmotorgui after absolute calibration, so that in this way there would have been no changes after switching to hard stop calibration. 94 or 93 degrees for hardware limits doesn't make any difference in practice, by the way it would be more correct to move the hardware limit to 93 in order to match the calibration value.

pattacini, 2022-03-02:
Thanks @S-Dafarra and @ale-git for the feedback 👍🏻 

>by the way it would be more correct to move the hardware limit to 93 in order to match the calibration value.

Agree on this 👍🏻 
@S-Dafarra, what about applying the tiny fix in your fork? The integration back on upstream would be easier this way.

S-Dafarra, 2022-03-03:
I was just checking the values. Right now, when calibrating the right leg, we expect the limit to be at 93.2 (see https://github.com/ami-iit/robots-configuration/blob/fab02fe65329fb7b30068e6586c8ab99f0c72a9b/iCubGenova09/calibrators/right_leg-calib.xml#L25). Is that 0.2 a problem of we set the hardware limit to 93? Would it make more sense to use 93.5, or just keep 94?

pattacini, 2022-03-03:
>Is that 0.2 a problem of we set the hardware limit to 93? Would it make more sense to use 93.5, or just keep 94?

I would call for @ale-git's educated opinion on that.