robotology/icub-models #138: Fail in configuration when ICUB_MODELS_COMPILE_PYTHON_BIN is enabled

Nicogene, 2022-03-07:
Not sure if it is due to #137 but if `ICUB_MODELS_COMPILE_PYTHON_BIN` is ON I get this error:

```
Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter Development
 CMake Error at bindings/CMakeLists.txt:8 (find_package):
   By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
   asked CMake to find a package configuration file provided by "pybind11",
   but CMake did not find one.

   Could not find a package configuration file provided by "pybind11" with any
   of the following names:

     pybind11Config.cmake
     pybind11-config.cmake

   Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
   "pybind11_DIR" to a directory containing one of the above files.  If
   "pybind11" provides a separate development package or SDK, be sure it has
   been installed.
```

cc @vvasco @AlexAntn

traversaro, 2022-03-07:
What is your configuration? How did you installed pybind11 and python? All robotology-superbuild jobs seem to be working fine : https://github.com/robotology/robotology-superbuild/runs/5442392415?check_suite_focus=true .

traversaro, 2022-03-07:
(If depending on your package manager you can provide the complete list of installed dependencies it would be great)

Nicogene, 2022-03-08:
This is my CMakeCache.txt:

```
# This is the CMakeCache file.
# For build in directory: /home/ngenesio/robotology/robotology-superbuild/build/src/icub-models
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a file.
ASSIMP_INCLUDE_DIRS:PATH=/usr/include

//Path to a library.
ASSIMP_LIBRARIES:FILEPATH=/usr/lib/x86_64-linux-gnu/libassimp.so

//Build libraries as shared as opposed to static
BUILD_SHARED_LIBS:BOOL=ON

Boost_DATE_TIME_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.71.0

//The directory containing a CMake configuration file for Boost.
Boost_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0

Boost_FILESYSTEM_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

Boost_IOSTREAMS_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.71.0

Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0

Boost_REGEX_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_regex.so.1.71.0

Boost_SYSTEM_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_system.so.1.71.0

Boost_THREAD_LIBRARY_RELEASE:STRING=/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0

//Path to a file.
CCD_INCLUDE_DIRS:PATH=/usr/include

//Path to a library.
CCD_LIBRARIES:FILEPATH=/usr/lib/x86_64-linux-gnu/libccd.so

//Path to a program.
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Initial cache
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//A wrapper around 'ar' adding the appropriate '--plugin' option
[... 1613 lines cut ...]
pkgcfg_lib_OGRE_PKGC_OgreMain-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_OGRE_PKGC_pthread
pkgcfg_lib_OGRE_PKGC_pthread-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_ASSIMP_assimp
pkgcfg_lib_PC_ASSIMP_assimp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_CCD_ccd
pkgcfg_lib_PC_CCD_ccd-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_CCD_m
pkgcfg_lib_PC_CCD_m-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_CURL_curl
pkgcfg_lib_PC_CURL_curl-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_fcl
pkgcfg_lib_PC_FCL_fcl-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_TINYXML2_tinyxml2
pkgcfg_lib_TINYXML2_tinyxml2-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_UUID_uuid
pkgcfg_lib_UUID_uuid-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAML_yaml
pkgcfg_lib_YAML_yaml-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ZIP_zip
pkgcfg_lib_ZIP_zip-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ZeroMQ_zmq
pkgcfg_lib_ZeroMQ_zmq-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu
```

Nicogene, 2022-03-08:
As you can see I didn't have `pybind11-dev`, after installing it, it configures without problems

traversaro, 2022-03-08:
Ack, I see. At the superbuild level that dependency was documented in https://github.com/robotology/robotology-superbuild/blob/master/doc/cmake-options.md#python . Anyhow, do you think we can close the issue?

Nicogene, 2022-03-08:
Yes sure, thanks!

closed 2022-03-08 by Nicogene