robotology/icub-models #152: Strange IMU Behaviour in Simulation

SimoneMic, 2022-04-14:
Hello,

Using stickBot in simulation, reading straight from the port /icubSim/chest/inertials/measures:o I have noticed some things that I can't explain.
Example of a message with the robot staying still:

```
(((-0.000261648547432980632665 0.000386437616020565775487 -6.33732246059674829246e-06) 4221.79699999999957072)) 
(((0.0134758698668982629387 -9.79686681490665201011 0.245724967233435981351) 4221.79699999999957072)) 
(((0.0 0.0 0.0) 4221.79699999999957072)) 
(((1.44258615603545758965 -4.1081595871458080893 -0.0789977909900776154029) 4221.79699999999957072)) 
() () () () () ()
```

First triplet should be orientation RPY angles, the second one is for accelerations, then gyroscope and magnetometer.

1. The gyroscope readings are all zero, and they don't change even when the robot is moving.
2. The gravity acceleration is on the Y axis.

Furthermore, when I command the robot to turn on itself, the orientation on the yaw angle follow a sinusoid with a different period and amplitude. The following graph is a comparison with the ground truth yaw orientation (from gazebo).
[image]

Blue is the ground truth.

Note: I have changed the IMU placement in the urdf file, to make it follow the ENU convention.

I have also tried reading the same port from `iCubGazeboV3` and here's a message-example:
```
(((-0.261571865089478661215 0.0103527618993429562644 0.0925871270225435000212) 49.5609999999999999432)) 
(((-0.021120326170310725622 -9.85860727378023504741 -0.055132951025752760521) 49.5609999999999999432)) 
(((0.0 0.0 0.0) 49.5609999999999999432)) 
(((-0.287991230245835116985 -0.00200742194989312955009 -0.00473901183109172625951) 49.5609999999999999432)) () () () () () ()
```

traversaro, 2022-04-14:
> First triplet should be orientation RPY angles, the second one is for accelerations, then gyroscope and magnetometer.

Are you sure about this? From https://github.com/robotology/yarp/blob/master/src/devices/multipleAnalogSensorsMsgs/multipleAnalogSensorsSerializations.thrift it seems that the order is:

* ThreeAxisGyroscopes;
* ThreeAxisLinearAccelerometers;
* ThreeAxisMagnetometers;
* OrientationSensors;
* <...>

> The gravity acceleration is on the Y axis.

Are you sure that this is not related to:

> Note: I have changed the IMU placement in the urdf file, to make it follow the ENU convention.

Can you share the modifications done in the model? Without this modifications the problem is the same?

SimoneMic, 2022-04-14:
> Can you share the modifications done in the model? Without this modifications the problem is the same?

For the orientation, after the modifications the acceleration is correct: without any modifications you have the messages present in the issue. After the change I have the gravity acceleration on the Z axes. Here what I have changed (in the comment the original):
```
<gazebo reference="chest">
    <sensor name="chest_imu_acc_1x1" type="imu">
      <always_on>1</always_on>
      <update_rate>100</update_rate>
      <!--<pose>0.0894115 0.0175 0.09929950000000001 -1.5707963267948968 -0.0 -1.5707963267948968</pose> -->
      <pose>0.0894115 0.0175 0.09929950000000001 0.0 0.0 -1.5707963267948968</pose>
      <plugin name="iCub_yarp_gazebo_plugin_IMU" filename="libgazebo_yarp_imu.so">
        <yarpConfigurationString>(yarpDeviceName chest-inertial) (disableImplicitNetworkWrapper)</yarpConfigurationString>
      </plugin>
    </sensor>
  </gazebo>
```
For the order I looked at the YARP wiki, but the only thing that I could find was from https://www.yarp.it/git-master/yarpdev.html to https://www.yarp.it/git-master/classServerInertial.html (although deprecated, I thought that the order could be the same, my bad)

SimoneMic, 2022-08-04:
Closing, since the IMU has a different orientation from the link frame

closed 2022-08-04 by SimoneMic