robotology/icub-models #156: iCubGazeboV2_* has `r_wrist_yaw` inverted

Nicogene, 2022-05-18:
As you can see from these videos, the rotation of `r_wrist_yaw` is the opposite w.r.t `l_wrist_yaw`:

https://user-images.githubusercontent.com/19833605/169014610-fa15450f-636d-41b3-8234-daf6549a2e41.mp4

On the other hand, on iCubGazeboV3, `iCubGazeboV2_5_visuomanip` and on the real robot left and right are moving accordingly:

iCubGazeboV3:

https://user-images.githubusercontent.com/19833605/169014672-a2e042c7-4540-4d8e-9d36-ca406b0c1e07.mp4

iCubGazeboV2_5_visuomanip:

https://user-images.githubusercontent.com/19152494/169041245-c7c3164e-9f62-4b86-916b-ee90ec0e94f9.mp4

traversaro, 2022-05-27:
Thanks @Nicogene . I think this is an issue similar to https://github.com/robotology/icub-models/issues/70 , but for iCubV2 . Whenever the Simmechanics model is regenerated from the Creo model, the actual direction of the exported axis is non-deterministic (from what we are able to understand in this years), and so the axis direction should be carefully checked and in case it is necessary we should update the `reverseRotationAxis` (https://github.com/robotology/icub-models-generator/blob/872c9bddaa33d1d3010ef1e63eb98760ebd9696d/simmechanics/CMakeLists.txt#L191). Apparently this went unnoticed, even because the tests for iCubV2 only involve the torso axis (see https://github.com/robotology/icub-models-generator/blob/0b244da996b1f0f308e6596d7e156c176e79c918/tests/icub-model-test.cpp#L227). 

By looking a bit at the history of generation, the last generation of the iCubV2 model was back in https://github.com/robotology/icub-models-generator/pull/122, so probably this bug was there in the last ~2.5 years, i.e. since icub-models 1.13.0 https://github.com/robotology/icub-models/releases/tag/v1.13.0 .

traversaro, 2022-05-30:
Mentioning @robotology/iit-artificial-mechanical-intelligence as it can be of interested to anyone working with `iCub**V2**` models: on those models the `r_wrist_yaw` joint has the wrong direction of motion since 2019.

Nicogene, 2022-05-31:
I can take this issue and work on it on spare time since it should not be complicated 👍🏻

traversaro, 2022-05-31:
Great, thanks!

closed 2022-06-01 by traversaro