robotology/icub-models #157: iCubGazeboV3 - joint limits of shoulder pitch not aligned with limits of iCubGenova09

isorrentino, 2022-05-25:
As you can see from this picture, the joint limits of the shoulder pitch joints of iCubGazeboV3 (all the models available for simulation) are not aligned with the ones of iCubGenova09.

[image]

These are the old limits, before the hardware modification and maybe the iCubGazeboV3 models have not been updated.

pattacini, 2022-05-25:
Hi @isorrentino 

We are about to release in the upcoming days an update of the iCub 3 URDF to fix some weight discrepancies.
We can take care of this as well.

Mick3Lozzo, 2022-06-29:
@isorrentino this will soon be fixed by @mfussi66 
cc @pattacini

pattacini, 2022-07-04:
@mfussi66 reported that:

>The model just pushed by the CI in `icub-models/devel` works as expected:
[image]
>
>And with the `/left_leg/analog:o` port working correctly to read the FT sensor.

The fix landed via https://github.com/robotology/icub-models/commit/8ad63f19211c3e8802ebb9f8593283713031dca9.

pattacini, 2022-07-04:
Closing, feel free to open it up again if deemed required.

closed 2022-07-04 by pattacini