robotology/icub-models #163: Load ICub model URDF files in CoppeliaSim

diegomaureira, 2022-06-21:
Hello! I'm developing a project with CoppeliaSim and I need to import an iCub model into the simulator but I haven't been able to do it until now. In this link https://drive.google.com/drive/folders/1A-ZVaEDVDF36tMxV0H2BDwl_SRrDTlZx?usp=sharing I show you the error messages that I've received when I've tried to import the URDF model of iCub Gazebo v2.7

traversaro, 2022-06-22:
I am not sure. In my system with ROS1, if I run:
~~~
(onlyros) traversaro@IITICUBLAP257:~/icub-models/build$ rospack find iCub
~~~
the output is:
~~~
/home/traversaro/icub-models/build/iCub
~~~

And then `/home/traversaro/icub-models/build/iCub/meshes/simmechanics/sim_sea_2-5_r_elbow_prt-binary.stl` is a valid mesh. Are you sure that you are launching `CoppeliaSim` in an environment in which `ROS_PACKAGE_PATH` contains `/home/diego/icub-models/build` ?

traversaro, 2022-06-24:
Great! How did you solved? Just to avoid a situation like https://xkcd.com/979/ . :D

diegomaureira, 2022-06-24:
Hello again! 
1. As you said: "rospack failed : This error seems to suggest that you did not defined the ROS_PACKAGE_PATH environment variable as described in the readme. This is probably the reason why you do not see meshes". I modify correctly ROS_PACKAGE_PATH and it works.
2. Regard to error message: "simExtURDF: error: neither mesh file 'iCub/meshes/simmechanics/sim_sea_2-5_r_elbow_prosup_prt-binary.stl' nor '' do exist". I raised the same thread in the coppelia forum and the admin's answer solved my problem, here I put you the link of the thread: https://forum.coppeliarobotics.com/viewtopic.php?p=37707&hilit=icub#p37707 

Cheers

traversaro, 2022-06-24:
Great, thanks. If you think the issue is solved, feel free to close it.

closed 2022-06-26 by diegomaureira