robotology/icub-models #164: Release new version of icub-models and whole-body-estimators

traversaro, 2022-07-29:
Apparently there is a strict relation between the recent changes in the iCub3 model (https://github.com/robotology/icub-models-generator/pull/223, https://github.com/robotology/icub-models/commit/8ad63f19211c3e8802ebb9f8593283713031dca9) and similar changes in the whole-body-estimators configuration files (see https://github.com/robotology/whole-body-estimators/pull/154).

To avoid confusion, it would be a good idea to make a new release of icub-models and whole-body-estimators, so that the latest version of both will be compatible with each other.

cc @RiccardoGrieco @RiccardoZuppetti @isorrentino @GiulioRomualdi @S-Dafarra

traversaro, 2022-07-29:
Indeed, @RiccardoZuppetti confirmed that only `devel` of `icub-models` is compatible with `master` of `whole-body-estimators` . While doing a release of `whole-body-estimators` requires coordination, something that we can already do to reduce friction for users is to `merge` devel in master for `icub-models`, to ensure that `master` of icub-models is compatible with `master` of `whole-body-estimators`.

traversaro, 2022-08-03:
I forgot to mention @HosameldinMohamed .

HosameldinMohamed, 2022-08-23:
Released [v0.8.0](https://github.com/robotology/whole-body-estimators/releases/tag/v0.8.0) for `whole-body-estimators`. to include https://github.com/robotology/whole-body-estimators/pull/154.

traversaro, 2022-08-23:
icub-models release: https://github.com/robotology/icub-models/releases/tag/v1.25.0 .

closed 2022-08-23 by traversaro