robotology/icub-models #169: Create iCubGazeboV3_visuomanip

traversaro, 2022-08-23:
For simulations related to teleoperation, it would be useful to have an iCubGazeboV3 model with moving eyes, cameras and moving hands. However, in the short term it would be difficult to obtain this information directly from the CAD, so probably we need to create this model by taking the hands and he head from the `[iCubGazeboV2_5_visuomanip](https://github.com/robotology/icub-models/tree/master/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip)`, and adding them to the existing `iCubGazeboV3` model. 

Related issues:
* https://github.com/robotology/icub-models/pull/42
* https://github.com/robotology/icub-models-generator/issues/37
* https://github.com/robotology/icub-models-generator/issues/29

cc @S-Dafarra

mebbaid, 2022-08-23:
On it, will look at the provided link, and maybe ask for help if needed.

Woolfrey, 2022-10-13:
Hi @traversaro, @mebbaid, I tried using this model and there's a bit of a problem with the hands!

[image]

Gazebo also seems to freeze a lot when I load this model, which may or may not be related to the detached hands:

[image]

mebbaid, 2022-10-13:
@Woolfrey : try to pull and compile against this fork https://github.com/mebbaid/icub-models instead of the usual `icub-models` master and see if this issue is resolved.

traversaro, 2022-10-13:
@mebbaid if that works, can you propagate it to https://github.com/robotology/icub-models/ ? Thanks!

mebbaid, 2022-10-13:
certo @traversaro . It seems to fix the issue on my side and Ehsan's side. Waiting fro @Woolfrey to confirm that the fix works for him as well before opening a PR.

Woolfrey, 2022-10-14:
@mebbaid , yes that seemed to work:

[image]

Woolfrey, 2022-10-17:
@mebbaid @traversaro, I got the model to load and it seems to be working with my controller, however it seems the right hand is producing some sort of force field when I push a box:

[image]

mebbaid, 2022-10-17:
This suggests to me that there is some issue with the `collision` parameters in the urdf for the right hand only. I will check it, and if that is the case I will push the fix to the fork

traversaro, 2022-10-17:
Note that in Gazebo you can enable the visualization of collision meshes to debug this sort of behaviour.

mebbaid, 2022-10-17:
Indeed it was the `collision` group of the `r_hand` link, as the following pic shows. 

[image]

I have pushed a commit https://github.com/mebbaid/icub-models/commit/0f25a16835a562c5091a2e3eb38dd8271e0aedbd in the fork with a fix as in the pic below. Let me know @Woolfrey if it works for you and that the intuition was correct.

[image]

Woolfrey, 2022-10-18:
@mebbaid, it seems to have worked!

[image]

PR #176 (2022-12-13): Fix minor issues in `iCubGazeboV3_visuomanip`
This is a PR that fixes
- `r_hand` collision parameters. reference issue #169 
- eyes pan joints poses. reference issue #https://github.com/ami-iit/component_ANA-Avatar-XPRIZE/issues/726
- eyes camera sensors and corresponding configurations. reference issue #https://github.com/ami-iit/component_ANA-Avatar-XPRIZE/issues/726

When finialized, this PR should close #169
closed 2022-12-13 by traversaro