robotology/icub-models #183: Gazebo crashes when inserting the iCubGazeboV* models

S-Dafarra, 2022-12-20:
As soon as I try to insert iCubGazeboV2_5, Gazebo crashes badly, with no error messages. It happens the same also with iCubGazeboV3

Here the output
<details>

```
sdafarra@iiticublap260:~$ gazebo -slibgazebo_yarp_clock.so
[INFO] GazeboYarpClock loaded. Clock port will be  /clock
[INFO] |yarp.os.Port|/clock| Port /clock active at tcp://192.168.1.129:10002/
[INFO] |yarp.os.Port|/clock/rpc| Port /clock/rpc active at tcp://192.168.1.129:10003/
Resetting YARP clock to default
[DEBUG] |yarp.dev.PolyDriver|multipleanalogsensorsserver| Parameters are WRAPPER (device multipleanalogsensorsserver) (name "/icubSim/head/inertials") (period 10)
[INFO] |yarp.dev.PolyDriver|multipleanalogsensorsserver| Created wrapper <multipleanalogsensorsserver>. See C++ class MultipleAnalogSensorsServer for documentation.
[DEBUG] |yarp.dev.PolyDriver|inertial| Parameters are ADDITIONAL_WRAPPER (device inertial) (name "/icubSim/inertial") (period 0.0100000000000000002082)
[WARNING] |yarp.devices.ServerInertial| The 'inertial' device is deprecated in favour of 'multipleanalogsensorsremapper' + 'multipleanalogsensorsserver' + 'IMURosPublisher'.
[WARNING] |yarp.devices.ServerInertial| The old device is no longer supported, and it will be deprecated in YARP 3.7 and removed in YARP 4.
[WARNING] |yarp.devices.ServerInertial| Please update your scripts.
[INFO] |yarp.devices.ServerInertial| No ROS group found in config file ... skipping ROS initialization.
[INFO] |yarp.os.Port|/icubSim/inertial| Port /icubSim/inertial active at tcp://192.168.1.129:10004/
[INFO] |yarp.devices.ServerInertial| Server Inertial : no ROS initialization required
[INFO] |yarp.dev.PolyDriver|inertial| Created wrapper <inertial>. See C++ class ServerInertial for documentation.
[DEBUG] |yarp.dev.PolyDriver|gazebo_imu| Parameters are (device gazebo_imu) (sensorScopedName "default::iCub::head::head_imu_acc_1x1") (sensor_name head_imu_acc_1x1)
[INFO] |yarp.dev.PolyDriver|gazebo_imu| Created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO] |yarp.os.Port|/icubSim/head/inertials/measures:o| Port /icubSim/head/inertials/measures:o active at tcp://192.168.1.129:10005/
[INFO] |yarp.os.Port|/icubSim/head/inertials/rpc:o| Port /icubSim/head/inertials/rpc:o active at tcp://192.168.1.129:10006/
[INFO] |yarp.devices.ServerInertial| Starting server Inertial thread
[INFO] Registered YARP device with instance name: default::iCub::head::head_imu_acc_1x1::dummy
[DEBUG] |yarp.devices.ServerInertial| Writing an Inertial measurement.
[DEBUG] |yarp.dev.PolyDriver|analogServer| Parameters are (device analogServer) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName l_leg_ft_sensor) (name "/icubSim/left_leg/analog:o") (period 10)
[WARNING] |yarp.devices.AnalogWrapper| The 'AnalogWrapper' device is deprecated.
[WARNING] |yarp.devices.AnalogWrapper| Possible alternatives, depending on the specific type sensor data, are:
[WARNING] |yarp.devices.AnalogWrapper| 'MultipleAnalogSensorsRemapper`+`MultipleAnalogSensorsServer`, `PoseStampedRosPublisher`, `WrenchStampedRosPublisher`,`IMURosPublisher`,etc.
[WARNING] |yarp.devices.AnalogWrapper| The old device is no longer supported, and it will be deprecated in YARP 3.7 and removed in YARP 4.
[WARNING] |yarp.devices.AnalogWrapper| Please update your scripts.
[INFO] |yarp.devices.AnalogWrapper| No ROS group found in config file ... skipping ROS initialization.
[INFO] |yarp.os.Port|/icubSim/left_leg/analog:o/rpc:i| Port /icubSim/left_leg/analog:o/rpc:i active at tcp://192.168.1.129:10007/
[INFO] |yarp.devices.AnalogWrapper|  : no ROS initialization required
[INFO] |yarp.dev.PolyDriver|analogServer| Created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName l_leg_ft_sensor) (name "/icubSim/left_leg/analog:o") (period 10) (sensorScopedName "default::iCub::l_leg_ft_sensor::l_leg_ft_sensor") (sensor_name l_leg_ft_sensor)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] |yarp.os.Port|/icubSim/left_leg/analog:o| Port /icubSim/left_leg/analog:o active at tcp://192.168.1.129:10008/
[INFO] Registered YARP device with instance name: default::iCub::l_leg_ft_sensor::l_leg_ft_sensor::dummy
[DEBUG] |yarp.dev.PolyDriver|analogServer| Parameters are (device analogServer) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName l_foot_ft_sensor) (name "/icubSim/left_foot/analog:o") (period 10)
[WARNING] |yarp.devices.AnalogWrapper| The 'AnalogWrapper' device is deprecated.
[WARNING] |yarp.devices.AnalogWrapper| Possible alternatives, depending on the specific type sensor data, are:
[WARNING] |yarp.devices.AnalogWrapper| 'MultipleAnalogSensorsRemapper`+`MultipleAnalogSensorsServer`, `PoseStampedRosPublisher`, `WrenchStampedRosPublisher`,`IMURosPublisher`,etc.
[WARNING] |yarp.devices.AnalogWrapper| The old device is no longer supported, and it will be deprecated in YARP 3.7 and removed in YARP 4.
[WARNING] |yarp.devices.AnalogWrapper| Please update your scripts.
[INFO] |yarp.devices.AnalogWrapper| No ROS group found in config file ... skipping ROS initialization.
[INFO] |yarp.os.Port|/icubSim/left_foot/analog:o/rpc:i| Port /icubSim/left_foot/analog:o/rpc:i active at tcp://192.168.1.129:10009/
[INFO] |yarp.devices.AnalogWrapper|  : no ROS initialization required
[INFO] |yarp.dev.PolyDriver|analogServer| Created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName l_foot_ft_sensor) (name "/icubSim/left_foot/analog:o") (period 10) (sensorScopedName "default::iCub::l_foot_ft_sensor::l_foot_ft_sensor") (sensor_name l_foot_ft_sensor)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] |yarp.os.Port|/icubSim/left_foot/analog:o| Port /icubSim/left_foot/analog:o active at tcp://192.168.1.129:10010/
[INFO] Registered YARP device with instance name: default::iCub::l_foot_ft_sensor::l_foot_ft_sensor::dummy
[DEBUG] |yarp.dev.PolyDriver|analogServer| Parameters are (device analogServer) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName r_leg_ft_sensor) (name "/icubSim/right_leg/analog:o") (period 10)
[WARNING] |yarp.devices.AnalogWrapper| The 'AnalogWrapper' device is deprecated.
[WARNING] |yarp.devices.AnalogWrapper| Possible alternatives, depending on the specific type sensor data, are:
[WARNING] |yarp.devices.AnalogWrapper| 'MultipleAnalogSensorsRemapper`+`MultipleAnalogSensorsServer`, `PoseStampedRosPublisher`, `WrenchStampedRosPublisher`,`IMURosPublisher`,etc.
[WARNING] |yarp.devices.AnalogWrapper| The old device is no longer supported, and it will be deprecated in YARP 3.7 and removed in YARP 4.
[WARNING] |yarp.devices.AnalogWrapper| Please update your scripts.
[INFO] |yarp.devices.AnalogWrapper| No ROS group found in config file ... skipping ROS initialization.
[INFO] |yarp.os.Port|/icubSim/right_leg/analog:o/rpc:i| Port /icubSim/right_leg/analog:o/rpc:i active at tcp://192.168.1.129:10011/
[INFO] |yarp.devices.AnalogWrapper|  : no ROS initialization required
[INFO] |yarp.dev.PolyDriver|analogServer| Created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (gazeboYarpPluginsRobotName icubSim) (gazeboYarpPluginsSensorName r_leg_ft_sensor) (name "/icubSim/right_leg/analog:o") (period 10) (sensorScopedName "default::iCub::r_leg_ft_sensor::r_leg_ft_sensor") (sensor_name r_leg_ft_sensor)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] |yarp.os.Port|/icubSim/right_leg/analog:o| Port /icubSim/right_leg/analog:o active at tcp://192.168.1.129:10012/
[INFO] Registered YARP device with instance name: default::iCub::r_leg_ft_sensor::r_leg_ft_sensor::dummy
[... 330 lines cut ...]
## finals_rebased
--------------------------------------------
/home/sdafarra/Software/robotology-superbuild/src/yarp-devices-forcetorque/: master
[... cut ...]
--------------------------------------------
/home/sdafarra/Software/robotology-superbuild/src/YARP_telemetry/: HEAD
8 months ago|Update CMakeLists.txt|fe8e77b08ebea2e282a047e53de631fbef8a70ec
## HEAD (no branch)
--------------------------------------------
/home/sdafarra/Software/robotology-superbuild/src/YCM/: ycm-0.14
5 months ago|Merge pull request #416 from diegoferigo/patch-2|c609e375bfec8afbf2f0413c0f162df930a88cd8
## ycm-0.14...origin/ycm-0.14
--------------------------------------------
```

</details>

traversaro, 2022-12-20:
Just to double check, can you try without `-slibgazebo_yarp_clock.so` ?

S-Dafarra, 2022-12-20:
> Just to double check, can you try without `-slibgazebo_yarp_clock.so` ?

Still crashing

S-Dafarra, 2022-12-20:
I have reinstalled the whole superbuild and the problem disappeared. Before I just tried to remove the Gazebo yarp plugins and icub-models build folder and the whole install folder. I guess there were other build folders I had to delete

S-Dafarra, 2022-12-20:
Actually, with ``iCubGazeboV3``, it crashed after about one minute. Here the output
<details>

```
sdafarra@iiticublap260:~$ gazebo -slibgazebo_yarp_clock.so
[INFO] GazeboYarpClock loaded. Clock port will be  /clock
[INFO] |yarp.os.Port|/clock| Port /clock active at tcp://192.168.1.129:10002/
[INFO] |yarp.os.Port|/clock/rpc| Port /clock/rpc active at tcp://192.168.1.129:10003/
Resetting YARP clock to default
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpDepthCamera| CC: m_driverParameters:	 (CAMERA_PARAM (focalLengthX 570.34222412109375) (focalLengthY 570.34222412109375) (tangentialPointX 0.0) (tangentialPointY 0.0)) (device gazebo_depthCamera) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName realsense_chest_depth) (sensorScopedName "default::iCub::chest::realsense_chest_depth") (yarpDeviceName icub_depth_camera)
[DEBUG] |yarp.dev.PolyDriver|gazebo_depthCamera| Parameters are (CAMERA_PARAM (focalLengthX 570.34222412109375) (focalLengthY 570.34222412109375) (tangentialPointX 0.0) (tangentialPointY 0.0)) (device gazebo_depthCamera) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName realsense_chest_depth) (sensorScopedName "default::iCub::chest::realsense_chest_depth") (yarpDeviceName icub_depth_camera)
[INFO] |yarp.dev.PolyDriver|gazebo_depthCamera| Created device <gazebo_depthCamera>. See C++ class GazeboYarpDepthCameraDriver for documentation.
[INFO] |gazebo-yarp-plugins.plugins.GazeboYarpDepthCamera| Registered YARP device with instance name: default::iCub::chest::realsense_chest_depth::icub_depth_camera
[DEBUG] GazeboYarpCamera Plugin: sensor scoped name is  default::iCub::chest::realsense_chest_rgb
[DEBUG] |yarp.dev.PolyDriver|gazebo_camera| Parameters are (device gazebo_camera) (disableImplicitNetworkWrapper) (framerate 30) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName realsense_chest_rgb) (sensorScopedName "default::iCub::chest::realsense_chest_rgb") (stamp 1) (yarpDeviceName icub_rgb_camera)
[INFO] |yarp.dev.PolyDriver|gazebo_camera| Created device <gazebo_camera>. See C++ class GazeboYarpCameraDriver for documentation.
[INFO] Loaded GazeboYarpCamera Plugin correctly
[INFO] GazeboYarpCamera: Register YARP device with instance name: default::iCub::chest::realsense_chest_rgb::icub_rgb_camera
[DEBUG] |yarp.dev.PolyDriver|gazebo_imu| Parameters are (device gazebo_imu) (sensorScopedName "default::iCub::chest::chest_imu_acc_1x1") (sensor_name chest_imu_acc_1x1)
[INFO] |yarp.dev.PolyDriver|gazebo_imu| Created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO] Registered YARP device with instance name: default::iCub::chest::chest_imu_acc_1x1::chest_inertial_hardware_device
[DEBUG] |yarp.dev.PolyDriver|gazebo_imu| Parameters are (device gazebo_imu) (sensorScopedName "default::iCub::head::head_imu_acc_1x1") (sensor_name head_imu_acc_1x1)
[INFO] |yarp.dev.PolyDriver|gazebo_imu| Created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO] Registered YARP device with instance name: default::iCub::head::head_imu_acc_1x1::head_inertial_hardware_device
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName l_leg_ft_sensor) (sensorScopedName "default::iCub::l_leg_ft_sensor::l_leg_ft_sensor") (sensor_name l_leg_ft_sensor) (yarpDeviceName icub_left_leg_ft)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] Registered YARP device with instance name: default::iCub::l_leg_ft_sensor::l_leg_ft_sensor::icub_left_leg_ft
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName l_foot_front_ft_sensor) (sensorScopedName "default::iCub::l_foot_front_ft_sensor::l_foot_front_ft_sensor") (sensor_name l_foot_front_ft_sensor) (yarpDeviceName icub_left_foot_front_ft)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] Registered YARP device with instance name: default::iCub::l_foot_front_ft_sensor::l_foot_front_ft_sensor::icub_left_foot_front_ft
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName l_foot_rear_ft_sensor) (sensorScopedName "default::iCub::l_foot_rear_ft_sensor::l_foot_rear_ft_sensor") (sensor_name l_foot_rear_ft_sensor) (yarpDeviceName icub_left_foot_rear_ft)
[INFO] |yarp.dev.PolyDriver|gazebo_forcetorque| Created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
[INFO] Registered YARP device with instance name: default::iCub::l_foot_rear_ft_sensor::l_foot_rear_ft_sensor::icub_left_foot_rear_ft
[DEBUG] |yarp.dev.PolyDriver|gazebo_forcetorque| Parameters are (device gazebo_forcetorque) (disableImplicitNetworkWrapper) (gazeboYarpPluginsRobotName iCub) (gazeboYarpPluginsSensorName r_leg_ft_sensor) (sensorScopedName "default::iCub::r_leg_ft_sensor::r_leg_ft_sensor") (sensor_name r_leg_ft_sensor) (yarpDeviceName icub_right_leg_ft)
[... cut ...]
[INFO] Closing device right_leg-mc_remapper
[INFO] Closing device left_leg-mc_remapper
[INFO] Closing device right_arm-mc_remapper
[INFO] Closing device left_arm-mc_remapper
[INFO] Closing device torso-mc_nws_yarp
[INFO] Closing device head-mc_nws_yarp
[INFO] Closing device right_leg-mc_nws_yarp
[INFO] Closing device left_leg-mc_nws_yarp
[INFO] Closing device right_arm-mc_nws_yarp
[INFO] Closing device left_arm-mc_nws_yarp
[INFO] shutdown phase finished.
corrupted size vs. prev_size in fastbins

```

</details>

S-Dafarra, 2022-12-20:
> Actually, with `iCubGazeboV3`, it crashed after about one minute. Here the output

This happened after I switched only ``icub-models`` to master