robotology/icub-models #185: Old configuration files - presumably not used

xEnVrE, 2023-02-14:
I noticed that there are several files inside https://github.com/robotology/icub-models/tree/devel/iCub/conf, where ideally only automatically generated file should stay, that are not updated by the GitHub action.

I am wondering whether these are really used by some of the automatically generated models or not. As an example, I think that automatically generated models are using the file:
- https://github.com/robotology/icub-models/blob/devel/iCub/conf/gazebo_icub_left_arm_no_hand_for_no_hand_model.ini

while the repository still hosts:
- https://github.com/robotology/icub-models/blob/devel/iCub/conf/gazebo_icub_left_arm_no_hand.ini
- https://github.com/robotology/icub-models/blob/devel/iCub/conf/gazebo_icub_left_arm_no_forearm.ini

having their last commit 3 years ago and still using the `controlboardwrapper2`.

Other files are probably not required anymore are:
- all the files having names like `gazebo_icub_{left, right}_hand_<finger_name>.ini` as the manual models have their equivalent counterparts inside https://github.com/robotology/icub-models/tree/devel/iCub_manual

Should we think of removing these files in order to tidy up the configuration folder?

traversaro, 2023-02-14:
Thanks for opening this issue! 

To avoid having this problem again, we should delete everytime all the files in the directory of automatically generated files.

traversaro, 2023-02-14:
Ok, I started tackling this with:
* https://github.com/robotology/icub-models/pull/186
* https://github.com/robotology/icub-models-generator/pull/236

Once this are done, we need to actually delete the files.

closed 2023-09-15 by traversaro